f4exb/sdrangel #4: Issue building, cmake, QOpenGLWidget

haxorthematrix, 2016-03-29:
I've followed the most recent build instructions in the Readme.md for Ubuntu 14.04, with an apt-get update && apt-get upgrade performed this morning and most recent clone form master about 30 minutes ago. The following is output from the "mkdir build && cd build && cmake ../ && make" command, which fails on the cmake.  I'm at a loss on the missing QOpenGLWidget, and the almighty Google is failing me.
I'd guess that I'm missing a dependency somewhere but all of the ones listed check out and are installed and up to date.

cmake fail:

Generating moc_syncmessenger.cpp
[  1%] Built target sdrbase_automoc
[  1%] Generating qrc_res.cpp
[  2%] Generating ui_mainwindow.h
[  3%] Generating ui_aboutdialog.h
[  3%] Generating ui_addpresetdialog.h
[  4%] Generating ui_basicchannelsettingswidget.h
[  4%] Generating ui_glscopegui.h
[  5%] Generating ui_glspectrumgui.h
[  5%] Generating ui_pluginsdialog.h
[  6%] Generating ui_preferencesdialog.h
Scanning dependencies of target sdrbase
[  7%] Building CXX object CMakeFiles/sdrbase.dir/sdrbase/mainwindow.cpp.o
In file included from /home/lpesce/sdrangel/sdrbase/mainwindow.cpp:46:0:
/home/lpesce/sdrangel/sdrbase/gui/glspectrum.h:24:25: fatal error: QOpenGLWidget: No such file or directory
 #include <QOpenGLWidget>

f4exb, 2016-04-12:
It is possible that QOpenGLWidget appears only in recent versions of Qt5. It was there on Qt 5.4 but may be not earlier. Anyway there should not be any need for it since it is a leftover from a failed attempt at migrating from QGLWidget. It has simply be removed and is presently of the fix branch. It will be moved progressively to other branches.

Thanks for reporting!

f4exb, 2016-04-24:
Fixed in release 1.2.0

closed 2016-04-24 by f4exb

nostromo123, 2016-12-06:
Have a vbox guest DEBIAN Jessie and Stretch 64bit on my PCLinuxOS host.

Tried to do an installation of the of the librepilot GUI on both

STRETCH seems to work, JESSIE terminates!

Only for information

`DEBIAN Stretch

user@nostromo:~$ sudo apt-get install build-essential ccache debhelper git-core git-doc flex graphviz bison libudev-dev libusb-1.0-0-dev libsdl1.2-dev python libopenscenegraph-dev qt5-default qttools5-dev-tools libqt5svg5-dev qtdeclarative5-dev qml-module-qtquick-controls libqt5serialport5-dev qtmultimedia5-dev qtscript5-dev libqt5opengl5-dev qml-module-qtquick-controls qml-module-qtquick-dialogs qml-module-qtquick-xmllistmodel qml-module-qtquick-localstorage qml-module-qtquick-particles2 qml-module-qtquick-window2 qml-module-qtquick2 libosgearth-dev openscenegraph-plugin-osgearth

user@nostromo:~$ sudo apt-get install libc6-i386

user@nostromo:~$ mkdir -p ~/code

user@nostromo:~$ cd ~/code

user@nostromo:~$ git clone https://bitbucket.org/librepilot/librepilot.git

user@nostromo:~$ cd ~/code/librepilot

user@nostromo:~$ git checkout next

user@nostromo:~$ make gcs

user@nostromo:~/code/librepilot/build/librepilot-gcs_release/bin$ ./librepilot-gcs

DEBIAN Jessie 

.
.
.

user@nostromo:~$ make gcs

/home/user/code/librepilot/ground/gcs/src/libs/opmapcontrol/src/mapwidget/opmapwidget.cpp:33:25: fatal error: QOpenGLWidget: Datei oder Verzeichnis nicht gefunden
 #include <QOpenGLWidget>
                         ^
compilation terminated.
`