f4exb/sdrangel #5: Error during make in glscope.cpp file

wintzx, 2016-05-12:
Hi 
I followed the build and dependencies instructions on Linux Ubuntu and have some troubles with make steps. Issues are related to glspcope.cpp  file (master pulled this day 2016-05-12)
sdrangel-master/sdrbase/gui/glscope.cpp: In member function ‘virtual void GLScope::resizeGL(int, int)’:
sdrangel-master/sdrbase/gui/glscope.cpp:292:15: error: ‘class QOpenGLFunctions’ has no member named ‘glViewport’  glFunctions->glViewport(0, 0, width, height);

There are some other issues
class QOpenGLContext’ has no member named ‘isOpenGLES
class QOpenGLFunctions’ has no member named ‘glClearColor
class QOpenGLFunctions’ has no member named ‘glClear

In fact all these functions are declared in /usr/include/qt5/QtGui/qopenglfunctions_es2.h
I surely missed some configuration related to openGL 
Is there a way to fix it ?. 
Any kind of help will be appreciated.
Thank you
Best regards,

f4exb, 2016-05-12:
Qt doc for QOpenGLFunctions says about glViewport: "this function was introduced in Qt 5.3." Are you using an earlier version of Qt by any chance? My build is based on Qt 5.5.

wintzx, 2016-05-13:
Well I have to compile QT5.6.0 from source and then use
/usr/local/bin/cmake -DCMAKE_PREFIX_PATH=/usr/local/Qt-5.6.0/ ../
After that there were no issues about openGL
But have another one in hack rf
/plugins/samplesource/hackrf/hackrfinput.cpp: In member function ‘bool HackRFInput::applySettings(const HackRFSettings&, bool)’:
/home/bggl9256/Developpements/iot/sdrangel-master/plugins/samplesource/hackrf/hackrfinput.cpp:363:85: error: ‘hackrf_set_antenna_enable’ was not declared in this scope
    rc = (hackrf_error) hackrf_set_antenna_enable(m_dev, (m_settings.m_biasT ? 1 : 0));

f4exb, 2016-05-13:
Hello,

OK so it's a Qt level problem. I have to mention that in the Readme.

For HackRF I think it is also a problem with the revision of the libhackrf. In fact antenna power control appears in this commit: https://github.com/mossmann/hackrf/commit/7c3f6340ed8cff29eb3776f3993a092f0b1cdfaa

Which is tagged with 2015.07.2 so I suppose you need at least this version of libhakcrf.

Best regards.

wintzx, 2016-05-17:
Hi
Thank you for your answers. Finally got it working well after building libhackrf with the lattest snapshot.
Here is my final command line to archieve the sdrangel compilation
/usr/local/bin/cmake -DCMAKE_PREFIX_PATH=/usr/local/Qt-5.6.0/ ../ -DLIBHACKRF_LIBRARIES=/usr/local/lib/libhackrf.so -DLIBHACKRF_INCLUDE_DIR=/usr/local/libhackrf/include
Just to mention that I am running Linux Ubuntu 14.04 LTS
You can close this topic. This was not really an issue. 
Again thank you for your kindness and sorry for the generated noise.
Best regards,
Pat

closed 2016-05-17 by wintzx

f4exb, 2016-05-18:
In the front Readme.md of the coming 2.0.0 version (already in dev) I have made sure these prerequisites are mentioned.