f4exb/sdrangel #8: compile error .. please ..

SonyKucing, 2016-11-10:
when i compile got error when step compile target inputbladerf ..... 
this error is ... 
`[ 81%] Built target inputbladerf_automoc
Scanning dependencies of target inputbladerf
[ 81%] Building CXX object plugins/samplesource/bladerf/CMakeFiles/inputbladerf.dir/bladerfgui.cpp.o
In file included from /root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfinput.h:22:0,
                 from /root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.h:23,
                 from /root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:27:
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfsettings.h:38:2: error: ‘bladerf_xb200_path’ does not name a type
  bladerf_xb200_path m_xb200Path;
  ^~~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfsettings.h:39:2: error: ‘bladerf_xb200_filter’ does not name a type
  bladerf_xb200_filter m_xb200Filter;
  ^~~~~~~~~~~~~~~~~~~~
In file included from /root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:27:0:
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.h:67:42: error: ‘bladerf_xb200_path’ has not been declared
  unsigned int getXb200Index(bool xb_200, bladerf_xb200_path xb200Path, bladerf_xb200_filter xb200Filter);
                                          ^~~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.h:67:72: error: ‘bladerf_xb200_filter’ has not been declared
  unsigned int getXb200Index(bool xb_200, bladerf_xb200_path xb200Path, bladerf_xb200_filter xb200Filter);
                                                                        ^~~~~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp: In constructor ‘BladerfGui::BladerfGui(DeviceSourceAPI*, QWidget*)’:
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:43:40: error: ‘BLADERF_FREQUENCY_MIN_XB200’ was not declared in this scope
  ui->centerFrequency->setValueRange(7, BLADERF_FREQUENCY_MIN_XB200/1000, BLADERF_FREQUENCY_MAX/1000);
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp: In member function ‘void BladerfGui::displaySettings()’:
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:200:74: error: ‘struct BladeRFSettings’ has no member named ‘m_xb200Path’; did you mean ‘m_xb200’?
  ui->xb200->setCurrentIndex(getXb200Index(m_settings.m_xb200, m_settings.m_xb200Path, m_settings.m_xb200Filter));
                                                                          ^~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:200:98: error: ‘struct BladeRFSettings’ has no member named ‘m_xb200Filter’; did you mean ‘m_xb200’?
  ui->xb200->setCurrentIndex(getXb200Index(m_settings.m_xb200, m_settings.m_xb200Path, m_settings.m_xb200Filter));
                                                                                                  ^~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp: In member function ‘void BladerfGui::on_xb200_currentIndexChanged(int)’:
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:299:14: error: ‘struct BladeRFSettings’ has no member named ‘m_xb200Path’; did you mean ‘m_xb200’?
   m_settings.m_xb200Path = BLADERF_XB200_BYPASS;
              ^~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:299:28: error: ‘BLADERF_XB200_BYPASS’ was not declared in this scope
   m_settings.m_xb200Path = BLADERF_XB200_BYPASS;
                            ^~~~~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:304:14: error: ‘struct BladeRFSettings’ has no member named ‘m_xb200Path’; did you mean ‘m_xb200’?
   m_settings.m_xb200Path = BLADERF_XB200_MIX;
              ^~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:304:28: error: ‘BLADERF_XB200_MIX’ was not declared in this scope
   m_settings.m_xb200Path = BLADERF_XB200_MIX;
                            ^~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:305:14: error: ‘struct BladeRFSettings’ has no member named ‘m_xb200Filter’; did you mean ‘m_xb200’?
   m_settings.m_xb200Filter = BLADERF_XB200_AUTO_1DB;
              ^~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:305:30: error: ‘BLADERF_XB200_AUTO_1DB’ was not declared in this scope
   m_settings.m_xb200Filter = BLADERF_XB200_AUTO_1DB;
                              ^~~~~~~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:310:14: error: ‘struct BladeRFSettings’ has no member named ‘m_xb200Path’; did you mean ‘m_xb200’?
   m_settings.m_xb200Path = BLADERF_XB200_MIX;
              ^~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:310:28: error: ‘BLADERF_XB200_MIX’ was not declared in this scope
   m_settings.m_xb200Path = BLADERF_XB200_MIX;
                            ^~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:311:14: error: ‘struct BladeRFSettings’ has no member named ‘m_xb200Filter’; did you mean ‘m_xb200’?
   m_settings.m_xb200Filter = BLADERF_XB200_AUTO_3DB;
              ^~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:311:30: error: ‘BLADERF_XB200_AUTO_3DB’ was not declared in this scope
   m_settings.m_xb200Filter = BLADERF_XB200_AUTO_3DB;
                              ^~~~~~~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:316:14: error: ‘struct BladeRFSettings’ has no member named ‘m_xb200Path’; did you mean ‘m_xb200’?
   m_settings.m_xb200Path = BLADERF_XB200_MIX;
              ^~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:316:28: error: ‘BLADERF_XB200_MIX’ was not declared in this scope
   m_settings.m_xb200Path = BLADERF_XB200_MIX;
                            ^~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:317:14: error: ‘struct BladeRFSettings’ has no member named ‘m_xb200Filter’; did you mean ‘m_xb200’?
[... 57 lines cut ...]
                       ^~~~~~~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:436:28: error: ‘BLADERF_XB200_AUTO_3DB’ was not declared in this scope
    else if (xb200Filter == BLADERF_XB200_AUTO_3DB)
                            ^~~~~~~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:440:28: error: ‘BLADERF_XB200_CUSTOM’ was not declared in this scope
    else if (xb200Filter == BLADERF_XB200_CUSTOM)
                            ^~~~~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:444:28: error: ‘BLADERF_XB200_50M’ was not declared in this scope
    else if (xb200Filter == BLADERF_XB200_50M)
                            ^~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:448:28: error: ‘BLADERF_XB200_144M’ was not declared in this scope
    else if (xb200Filter == BLADERF_XB200_144M)
                            ^~~~~~~~~~~~~~~~~~
/root/GSM_hacking/sdrangel/plugins/samplesource/bladerf/bladerfgui.cpp:452:28: error: ‘BLADERF_XB200_222M’ was not declared in this scope
    else if (xb200Filter == BLADERF_XB200_222M)
                            ^~~~~~~~~~~~~~~~~~
plugins/samplesource/bladerf/CMakeFiles/inputbladerf.dir/build.make:66: recipe for target 'plugins/samplesource/bladerf/CMakeFiles/inputbladerf.dir/bladerfgui.cpp.o' failed
make[2]: *** [plugins/samplesource/bladerf/CMakeFiles/inputbladerf.dir/bladerfgui.cpp.o] Error 1
CMakeFiles/Makefile2:1637: recipe for target 'plugins/samplesource/bladerf/CMakeFiles/inputbladerf.dir/all' failed
make[1]: *** [plugins/samplesource/bladerf/CMakeFiles/inputbladerf.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
`

thanks ...

f4exb, 2016-11-11:
Hello,

it looks like you are using a very old version of libbladerf. You need to upgrade your installation with a more recent version. I am currently building with version 1.4.3. It should also work with the latest packaged release (see: https://github.com/Nuand/bladeRF/releases)

SonyKucing, 2016-11-11:
thanks master.. now all run perfect .....

f4exb, 2016-11-13:
Great, enjoy!

closed 2016-11-13 by f4exb