f4exb/sdrangel #9: No package 'libnanomsg' found

streamofstars, 2016-11-17:
Cannot compile due to the following error

```
-- Checking for module 'libnanomsg'
--   No package 'libnanomsg' found
-- libnanomsg not found.
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
/home/<myusername>/sdrangel/plugins/samplesource/sdrdaemon/LIBNANOMSG_INCLUDE_DIR
   used as include directory in directory /home/<myusername>/sdrangel/plugins/samplesource/sdrdaemon
LIBNANOMSG_LIBRARIES (ADVANCED)
    linked by target "inputsdrdaemon" in directory /home/<myusername>/sdrangel/plugins/samplesource/sdrdaemon

-- Configuring incomplete, errors occurred!
```

System details:
```
sh -c "uname -a;lsb_release -a"
Linux <myhostname> 4.4.0-47-generic #68-Ubuntu SMP Wed Oct 26 19:39:52 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.1 LTS
Release:	16.04
Codename:	xenial
```

I do not see libnanomsg listed as a prerequisite. Is [this](https://github.com/ZewoGraveyard/libnanomsg) the one I need to install?

SonyKucing, 2016-11-17:
apt-get install libnanomsg-dev libnanomsg4

streamofstars, 2016-11-17:
`E: Unable to locate package libnanomsg4`
However, installed only `libnanomsg-dev` and it compiled. Now it starts but nothing happens. It lists my rtlsdr dongle on the list of devices, however does not list my airspy mini. I cannot set the frequency and clicking on the play-icon button does nothing.

SonyKucing, 2016-11-18:
after SdrAngel program is opened, click on dropdown menu file source, change to your device sdr name is listed, then click on "tick" icons on right device list, that mean you confirm that correct device for you, and frequency will be can to change...  please try to surf that program by your self, trial and error, because nohting tutorial for use this program....

f4exb, 2016-11-21:
Please look for the tooltips as you mouse over buttons and other GUI features. I am sorry for the lack of tutorials but I am very busy with the development and my day job also. I have tried to add some readme files here and there to help understanding the interface. These readme's should be accessible directly from the Wiki.

As for the original subject I keep this ticket open as the full compilation should not be dependent on libnanomsg. This library is used only to compile the SDRDaemon and SDRdaemonFEC plugins. I think this does need a fix in the build process with cmake.

f4exb, 2016-11-23:
Fixed in release 2.3.0

closed 2016-11-23 by f4exb