f4exb/sdrangel #10: output from recorded signal

SonyKucing, 2016-12-08:
hy bro...  i need help about output file from recorded sdrangel, because i need to use on inspectrum .. 
`    *.cf32, *.cfile - Complex 32-bit floating point samples (GNURadio, osmocom_fft)
    *.cs16 - Complex 16-bit signed integer samples (BladeRF)
    *.cs8 - Complex 8-bit signed integer samples (HackRF)
    *.cu8 - Complex 8-bit unsigned integer samples (RTL-SDR)
`
thanks..

f4exb, 2016-12-09:
The sample format is the native sample format used in SDRangel which is 2 x 2 bytes signed (or short) values for a complex sample. In addition there is a header with this structure:
    struct Header
    {
        int         sampleRate;
        quint64     centerFrequency;
        std::time_t startTimeStamp;
    };

On most systems this will compile as 24 bytes i.e. 6 samples. So if you admit some glitch at the start (these 6 'false' samples) this is directly convertible from a 2 x short format to anything else using a conversion program like sox using two channels to take the complex structure into account. 

For GNURadio you can simply specify your file source format as short complex.

Multiple format support directly in SDRangel is not a priority knowing that there are workarounds as described above.

Note that you can also zap the 24 byte header with standard Linux utilities.

SonyKucing, 2016-12-09:
thanks bro...   (y)

f4exb, 2016-12-16:
Added comments in the file record section of the main display radme in sdrbase folder

closed 2016-12-16 by f4exb