f4exb/sdrangel #18: DATV: implement Digital Amateur TV modulator and demodulator plugins

f4exb, 2017-03-24:
All is in the title...

f4exb, 2017-12-28:
This is undertaken by a contributor not listed in this project.

f4exb, 2018-03-17:
Done in v3.13.0

closed 2018-03-17 by f4exb

darksidelemm, 2021-02-20:
Currently it looks like the DATV demodulator only provides channel power, not the more useful SNR or MER (as is usually used in DVB work). Being able to quantify the signal quality is pretty important when testing systems...
The leandvb demodualtor (at least the current revision of it ) does provide this - is it possible for this information to be exposed on the SDRangel DATV GUI?

f4exb, 2021-02-20:
The DATV demodulator is so bad anyway at least for DVB-S2 that before talking about any such improvement the demodulator itself should be improved first maybe by dropping leandvb that has pretty horrible C++ code making it necessary to rewrite parts. Therefore it cannot be upgraded easily.

f4exb, 2021-02-20:
QPSK demodulation seems to work correctly. Constellation synchronizes both with the Channel Analzyer PLL and DATV leansdr. So issus may be with decoder blocks:

[image]

darksidelemm, 2021-02-20:
I'm actually only using DVB-S at the moment :-) 

I'm launching a DVB-S transmitter on a high-altitude balloon in a few weeks (1Msps QPSK, r=1/2 FEC), and I've actually recommended SDRangel as the easiest way for many users to get setup to try and receive video from it using equipment they may already have. 
I'm quite interested in what SNR the various listeners observer, hence asking the question about a SNR metric display. I guess for now I'll have to get them to screenshot the constellation diagram, which can give a rough indication of SNR.

f4exb, 2021-02-21:
Yes DVB-S seems to work correctly. By SNR I suppose you mean CNR. I don't think leandvb has a SNR metric strictly speaking. I have seen in the code provision to get the CMR as an option but it does not seem to be fully implemented.

darksidelemm, 2021-02-21:
Yes I should have clarified - LeanDVB reports 'MER', which is related to SNR but apparently incorporates more information about the modulation. You can see it on the leandvb GUI here: https://twitter.com/vk5qi/status/1363026789761650688

f4exb, 2021-02-21:
Unfortunately modifying the code is very difficult because as mentioned in the leandvb documentation: http://www.pabr.org/radio/leandvb/leandvb.en.html#usage_ui leandvb is "not designed to be linked with other software". Normally it is compiled as a standalone program interacting with standard input / output so in the plugin it is used in a unsupported way.

darksidelemm, 2021-02-21:
Well.. the code to calculate the MER does appear to be in the fork of leansdr that's in sdrangel.

In the upstream leansdr code the place where the MER is printed is here: https://github.com/pabr/leansdr/blob/master/src/apps/leandvb.cc#L605

In your fork: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demoddatv/datvdemodsink.cpp#L666
though it doesn't look like it gets read from anywhere. 
It looks like you are reading off the p_sampled data for the constellation diagram. I guess the question is - what is required to read out the p_mer data and display that somewhere?

f4exb, 2021-02-21:
The `p_mer` pipe (`pipebuf`) is allocated but does not lead to anywhere. In the `main` program `leandvb` it is routed to a file via the `file_printer`. This is not what you want with SDRangel. In fact what has been done to show the constellation is subclassing `runnable` writing a custom leandvb "class" like this: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demoddatv/datvconstellation.h#L111 then it is integrated in the global pipe like this: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demoddatv/datvdemodsink.cpp#L764

darksidelemm, 2021-02-22:
This is great news! I see the p_mer line is commented out for the DVB-S2 demodulator, and as you say it doesn't perform that well - however the DVB-S demod seems to work just fine, and there are certainly still DVB-S users out there that would benefit from a MER report!

f4exb, 2021-02-22:
Now that MER and CNR can be displayed (it is a prototype) the limit for DVB-S2 to decode is something like this:
[image]

darksidelemm, 2021-02-22:
NIce work! This will work for 'classic' DVB-S too?

f4exb, 2021-02-23:
Yes. It is now on master. Occasionally there is an abort when closing the program. More likely on DVB-S2 than DVB-S.