f4exb/sdrangel #20: LimeSDR support

f4exb, 2017-03-31:
LimeSDR support with its native interface (not SoapySDR)

f4exb, 2017-04-10:
Sources of information:

Main interface is here:
https://github.com/myriadrf/LimeSuite/blob/master/src/lime/LimeSuite.h

Docs:
https://github.com/myriadrf/LimeSuite/tree/master/docs
https://github.com/myriadrf/LimeSuite/blob/master/docs/lms7_api_quick_start_guide.pdf

Examples:
https://github.com/myriadrf/LimeSuite/tree/master/src/examples

Examples are built in the bin subdirectory of the build directory and are not installed by default.

LMS7002M info:
http://wiki.myriadrf.org/LimeMicro%3ALMS7002M_Datasheet
https://github.com/myriadrf/LMS7002M-docs

f4exb, 2017-04-10:
Constraints particular to LimeSDR (LMS7002M): 
  - same base sample rate for Rx and Tx (but oversampling can be varied)
  - same center frequency for all Rx channels on one part and all Tx channels on the other part

f4exb, 2017-04-11:
Note that LimeSDR will not be supported in Windows due to too many errors trying to compile Lime Suite in mingw.

f4exb, 2017-04-20:
For operation on HF bands and digital non-zero IF we need to implement the NCO feature. The question is what are the frequency limits of this NCO. This is indirectly answered here: 

https://github.com/myriadrf/LimeSuite/issues/101

"a 1 Msps rate means that the ADC rate will be 32 Msps. So when the LO tunes down to 30 MHz, we can only downshift by an additional 16 MHz"

So it is simply +/- half the ADC actual rate which is the LMS<->Host sample rate multiplied by the hardware decimation/interpolation factor.

f4exb, 2017-04-24:
Published in release v3.4.0

closed 2017-04-24 by f4exb

DG9BFC, 2017-05-23:
lime suite runs fine in windows ... but i can not get sdrangel working :-(