f4exb/sdrangel #21: Full Analog-TV support

veso266, 2017-04-03:
Hi do you think you could add Full Analog TV Support (not just ATV Analog)

It would be great if you could add PAL B/G, D/K, I and NTSC as well as SECAM
but first if you could add Color support to existing PAL Demodulator (I have a lot of bosnian stations that are still analog and nowadays its imposible to find a portable Analog television that would support all analog television modes 

Thanks for Anwsering and Best Regards

f4exb, 2017-04-04:
I have added it as a feature request and realize that analog color support is of interest for some of the audience. However it is fairly demanding in development effort and is not my priority. As always I am opened to contributions.

Also it would help if the present analog BW ATV gets more mature. For example it is still weak on the line synchronization and this is a killer for color support.

veso266, 2017-04-05:
OK so when you finish all your issues and improve line sync will color be added
PS: don't hams like colors? (why would it be of interest to them?)(even PAL Plugin for SDR# stopped in black and white state) I am asking this because I don't understand why adding color is that demanding because if you came that far to actualy implement PAL isn't adding color just a small addition?

Thanks for Anwsering and Best Regards

f4exb, 2017-04-06:
When hams want to use color they probably switch do digital nowadays and some probably did that some time ago already. Purpose of the plugin was to have fun with good old analog TV without too many bells and whistles and also experiment with NBTV (Narrow Band) and for that purpose you would discard color to save bandwidth.

I used the term "PAL" to qualify the frame synchronization scheme which is the one used in PAL (and NTSC and SECAM by the way) and that's all that is PAL there. To implement color is a lot of work because you have to mimic the analog circuitry. This implies locking the color oscillator (a NCO then) in frequency and phase with the small burst at the start of the line. Then you need to filter the luminance (LPF) and the chrominance (BPF) apart. You then take the chrominance signal and decode its phase using the reference of the color oscillator. Then you reassemble everything to get the RGB according to the particular standard (PAL, NTSC, SECAM). I beleieve this is a lot of work for just analog color. Now anyone who is up at it is welcome but I will certainly concentrate on more valuable features. In particular it would be more interesting to add support for digital.

So unless a contribution comes in I would say that it is not anytime soon.

DG9BFC, 2017-05-23:
we also still use analog atv here ... so i second that begging for pal color support
would make an easy atv repeater
rx on 2.4 tx on 3,4 gig (just an example) ... or use lnb for 10gig rx and retransmit on 13 or 23 cm??
all is possible with the lime
(if i could just get sdrangel working here grmbl)

cabeener, 2017-10-18:
In the USA Hams on ATV still use analog TV NTSC. Some of us have filter at 70mhz to take the bandwidth down to 350Khz, 700Khz, 1.2 Mhz up to 3.5Mhz. of course when we use 350Khz the Ham Txing needs to use Large black Call letter so we can see them. It helps a lot. you can see Call letter about 12db down farther in the noise.
 A normal Rx sensitive for analog TV is around -100dbm but with 350khz its down around -115dbm.
You can take just seeing sync at 3.5Mhz to a lock P1 at 350khz bandwidth. 
  So I'm looking for better NTSC analog TV, and also DVB-T 2Mhz wide to 100Khz wide.
At 2mhz wide we have worked W4HTB two way Digital DVB-T 2Mhz. 311 miles just once.
But have work 125 miles all the time. 
  We love your program keep up the good work !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Charles
WB8LGA

DG9BFC, 2017-12-19:
our repeater has a dvbt receiver 2 megs wide on 432 megs ... works suerb even for long distances
cabeener what did you use ther for dvbt?? the stuff from taiwan??
the lime maybe also could do this (its all a question of software) ...
with a wide band multiband rx tx like the lime you could do ANY mode ... even dvbt or any other fancy digital mode ... in any bandwidth

f4exb, 2017-12-20:
Request for DATV modes is rather followed up with this issue https://github.com/f4exb/sdrangel/issues/18

DG9BFC, 2017-12-20:
analog tv is not obsolete so color support is a must have (i think) ... i was just asking cabaneer what he has used when they made their tests in datv ... i did not want to get off topic (and yes i have seen that topic)
so my dream is:
a powerfull single board pc with a touch screen (odroid xu4 with vu7 or a tablet pc) together with the lime and some poweramps, filters, rx/tx relays, gpio board, 5 and 10 gig up/downmixer all in one case
running sdrangel and we have a multiband multimode sdr transceiver
most analog things are already there (am fm ssb etc.)
on analog only analog tv is not finished .. when that is done and if dvbt could be added ... then also that mode could be added without adding new hardware tm my "dream" sdr :)

cabeener, 2017-12-20:
Yes we used Hides equipment for Tx and Rx.
But there is a new SDR out Adalm-Pluto. You can Tx DVB-T on it with program Express DVB Transmitter.
and It only cost $99 USD. We have work two way on DVB-T 2 Meg and 311 miles. and we do 125 mile.
on and off depending the conditions. Yes you can also use Lime on both Tx and Rx but not in Windows.

F5OEO, 2018-03-20:
Can maybe inspired on Tx by colour ATV transmitter from https://github.com/fsphil/hacktv

f4exb, 2019-03-24:
This issue has gained interest since the 5 GHz band FPV devices operate in color analog either PAL 625/25 or NTSC 525/30 with FM modulation. Using a modified version of hacktv (deviation for FPV is apparently only 4.3 MHz) I was able to see the color video on a wrist watch receiver. Great fun! 

I think putting color in the ATV transmitter first. This is easier than the Rx.

DG9BFC, 2019-03-24:
what about also adding tone carrier(s) ???
5.5-6.5mhz ... selectable in 500kc steps (thats all needed 5.5, 6, 6.5 megs carrier)
or free tunable ??
greetz sigi dg9bfc

f4exb, 2019-04-28:
Sorry, not enough resources this request is dropped

f4exb, 2019-04-29:
Deserves second look

f4exb, 2022-02-10:
dropped

PhilippMolitor, 2023-10-04:
Is there any chance this will be revived? I am coming from exactly the use case you described: I'm flying FPV and would love to use SDRAngel to view analog video feeds, while also using other SDR features (mostly just the spectrogram) in the 5.8G Band.

gllxflr, 2024-03-23:
Would love this to be revived, or at the very minimum a proper output from the existing one. 560x418 (viewport size) isn't a suitable capture resolution for 625-line television

DG9BFC, 2024-03-24:
I guess with clever firmware one could stream video to pluto and get full analog tx signal (with colour and tone-subcarrier!!!) we can send DVB-S/S2 and DVB-T with the Pluto ... so I guess the Pluto could send analog (in PAL resolution)... just needs a clever guy coding itDG9BFC sigi

srcejon, 2026-11-01:
Bit late, but...

[image]

srcejon, 2026-11-09:
Implemented in 8.1

closed 2026-11-09 by srcejon