f4exb/sdrangel #25: FM demodulator mangles bits destined for an external audio modem

g0hww, 2017-05-02:
I've recently started using sdrangel with my HackRF and have found that it doesn't seem able to feed FM demod audio into multimon-ng (via pulse audio) in a way that allows data to be decoded.

It looks like it might be AC coupled rather than DC coupled.  Here's a picture from mulimon-ng's SCOPE sink, showing the marks and spaces decaying to zero;
[image]

By contrast, here is a screenshot of the resultant demod audio from the same transmitter, this time demodulated by gqrx, in which the marks and spaces seem to be well shaped:
[image]

I wonder if maybe there's a filter for CTCSS tones that could be doing this, or if there is another aspect of the NB FM demodulator that is responsible.

BTW, I really like sdrangle.  Thanks for your efforts.
Cheers,
Darren, G0HWW

f4exb, 2017-05-04:
Hello,

firstly thanks for your interest in SDRangel.

The NFM (or NBFM) demodulator plugin is intended for audio and indeed it has a high-pass filter for CTCSS permanently engaged to remove possible sub-tones below 300 Hz that produce an annoying wobbling sound.

I think the NFM demod and pulseaudio combination is not the best combination suited for your application. Instead you should make it work via UDP using the "UDP channel source" plugin and in a terminal pipe netcat output into multimon-ng input.

I have successfully decoded APRS frames with the following settings:

In the UDP channel source plugin:
  - Format: S16LE NFM Mono
  - Rate: 22050 Hz
  - Addr: 127.0.0.1
  - RF BW: 15000 Hz
  - D: 9999 (this is the data port, choose whatever suits your installation)
  - A: 9998 (this is the port for audio return and is not used in this case)
  - FMd: 6000 Hz (this is the expected maximum FM deviation, you have to fiddle with it)
  - Boost: 0 and Volume: 20 (leave the defaults)

On the command line in a terminal:

<pre>
nc -l -u 9999 | multimon-ng -A -a SCOPE -
</pre>

Of course the port number must match the data port. The scope is convenient when adjusting FM deviation in the UDP plugin.

I hope this helps.

Best regards.

g0hww, 2017-05-04:
Aha!  Thanks for the informative answer. Your suggestion works perfectly.
I've been shoving stuff through pulseaudio for so long that I forgot there are better ways of handling data interchange between radios and modems :)

I did notice that the volume control in the UDP sink didn't seem to have any effect on the levels shown in the scope sink of multimon-ng, although the default level (20) seems fine, so I'll change the title of this issue to properly reflect the name of the NFM demod in sdrangel and I'm happy to close it.

73

closed 2017-05-04 by g0hww