f4exb/sdrangel #28: PulseAudio

KC7NOA, 2017-05-18:
either Pulseaudio is not used or not implemented ...

I could use Alsa but my first device is not used on an Odroid-x2. HardKernel cant seem to write an audio driver to save the company ... so in pulse i have it disabled. 

In my special circumstance -- hdmi audio is also impossible. My HDMI output is blown.

My default audio setting is 48Khz 16LE ....

Ubuntu16.04 32bit 

Ill start another issue because on the Odroid EGS is not quite right, display uses RGB and qt needs a more specific rendering type to display correctly.(this is something that had buged me for some time about HardKernel use of Mali400 GPU -- which i cant even use ,,,, i use a Displaylink165)

KC7NOA, 2017-05-21:
Just trying to get some insight

kc7noaodroid@odroid:~$ sdrangel

(process:4848): Gtk-WARNING **: Locale not supported by C library.
    Using the fallback 'C' locale.
QMetaObject::connectSlotsByName: No matching signal for on_sampleSource_confirmClicked(bool)
QMetaObject::connectSlotsByName: No matching signal for on_sampleSink_confirmClicked(bool)
QMetaObject::connectSlotsByName: No matching signal for on_channel_addClicked(bool)
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/libcodec2.so.0.5'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/libfcdhid.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/libfcdlib.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/libgnuradio-osmosdr.so.0.1.5git'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/libitpp.so.8.2.1'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/liblimesdrdevice.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/libLimeSuite.so.17.03.0'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/libmbe.so.1.0'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/libpsk.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/libsdrbase.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/libSoapySDR.so.0.6.0'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/libwdsp.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/kodi/addons/library.kodi.adsp/libKODI_adsp-arm.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/kodi/addons/library.kodi.audioengine/libKODI_audioengine-arm.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/kodi/addons/library.kodi.guilib/libKODI_guilib-arm.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/kodi/addons/library.xbmc.addon/libXBMC_addon-arm.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/kodi/addons/library.xbmc.codec/libXBMC_codec-arm.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/kodi/addons/library.xbmc.pvr/libXBMC_pvr-arm.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/kodi/system/ImageLib-arm.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/kodi/system/libcpluff-arm.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/kodi/system/libexif-arm.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/kodi/system/players/dvdplayer/libdvdcss-arm.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/kodi/system/players/dvdplayer/libdvdnav-arm.so'
PluginManager::loadPlugins: loaded plugin libchanalyzer.so
PluginManager::loadPlugins: loaded plugin libchanalyzerng.so
PluginManager::loadPlugins: loaded plugin libdemodam.so
PluginManager::loadPlugins: loaded plugin libdemodatv.so
PluginManager::loadPlugins: loaded plugin libdemodbfm.so
PluginManager::loadPlugins: loaded plugin libdemodlora.so
PluginManager::loadPlugins: loaded plugin libdemodnfm.so
PluginManager::loadPlugins: loaded plugin libdemodssb.so
PluginManager::loadPlugins: loaded plugin libdemodtcpsrc.so
PluginManager::loadPlugins: loaded plugin libdemodudpsrc.so
PluginManager::loadPlugins: loaded plugin libdemodwfm.so
PluginManager::loadPlugins: loaded plugin libmodam.so
PluginManager::loadPlugins: loaded plugin libmodatv.so
PluginManager::loadPlugins: loaded plugin libmodnfm.so
PluginManager::loadPlugins: loaded plugin libmodssb.so
PluginManager::loadPlugins: loaded plugin libmodwfm.so
PluginManager::loadPlugins: loaded plugin liboutputfilesink.so
PluginManager::loadPlugins: loaded plugin liboutputlimesdr.so
PluginManager::loadPlugins: loaded plugin libinputfcdpro.so
PluginManager::loadPlugins: loaded plugin libinputfcdproplus.so
PluginManager::loadPlugins: loaded plugin libinputfilesource.so
PluginManager::loadPlugins: loaded plugin libinputlimesdr.so
PluginManager::loadPlugins: loaded plugin libinputrtlsdr.so
PluginManager::loadPlugins: loaded plugin libinputsdrdaemon.so
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/_SoapySDR.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/osmosdr/_osmosdr_swig.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/quisk/_quisk.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/quisk/sdriqpkg/sdriq.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/xpra/codecs/argb/argb.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/xpra/codecs/csc_cython/colorspace_converter.so'
PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/xpra/codecs/enc_x264/encoder.so'
[... cut ...]
LimeSDRInput::applySettings: could not disable and set NCO to -24000000 Hz
############################################################
Rx calibration using RSSI INTERNAL ON BOARD loopback
Rx ch.A @   30 MHz, BW: 4.5 MHz, RF input: LNAW, PGA: 0, LNA: 8, TIA: 1
Performed by: MCU
------------------------------------------------------------
MCU algorithm time: 1 ms
Current MCU firmware: 3, DC/IQ calibration full
MCU Ref. clock: 30.72 MHz
MCU algorithm time: 249 ms
AudioOutput::start: using default device alsa_output.usb-0d8c_C-Media_USB_Headphone_Set-00.analog-stereo

f4exb, 2017-05-24:
Hello,

this is the problem:
"QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled" 

You DO need a working OpenGL on your system as a prerequisite to display the OpenGL parts which are the spectrum, scope and TV monitor windows.

Please make sure that OpenGL is working in your system first.

Best regards.

closed 2017-05-24 by f4exb

KC7NOA, 2017-05-30:
yes .. verified ... after searching some ... DisplayLink under linux DOES NOT have opengl support ... bummer

KC7NOA, 2019-01-27:
Displaylink, it turns out, has nothing to do with the issue ....

Quote from maveric "In fact if you have the correct Qt5 version it should already use GLES anyway."
page https://forum.odroid.com/viewtopic.php?f=136&t=31237&p=244557#p244557

with a picture of what i see using hdmi and a HD tv

KC7NOA, 2019-01-28:
this does seem to work ...
kc7noa@odroid:~$ glmark2
MESA-LOADER: failed to retrieve device information
libGL error: unable to load driver: mali_drm_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: mali_drm
** GLX does not support GLX_EXT_swap_control or GLX_MESA_swap_control!
** Failed to set swap interval. Results may be bounded above by refresh rate.
=======================================================
    glmark2 2014.03+git20150611.fa71af2d
=======================================================
    OpenGL Information
    GL_VENDOR:     VMware, Inc.
    GL_RENDERER:   llvmpipe (LLVM 6.0, 128 bits)
    GL_VERSION:    3.0 Mesa 18.0.5
=======================================================
** GLX does not support GLX_EXT_swap_control or GLX_MESA_swap_control!
** Failed to set swap interval. Results may be bounded above by refresh rate.
[build] use-vbo=false:
there is a spinning horse/unicorn figure