f4exb/sdrangel #29: just downloaded windows version ... but it does not work here :-(

DG9BFC, 2017-05-23:


f4exb, 2017-05-24:
Hello,

a working OpenGL installation is a must. Can you please check if OpenGL is correctly installed in your system following the procedure described here; https://blogs.esri.com/esri/supportcenter/2012/03/15/what-version-of-opengl-do-i-have-and-do-i-have-the-latest-drivers-for-my-system/

Best regards,

DG9BFC, 2017-05-24:
testtool reports open gl works (i have seen that turning cube in the test)
opengl version 2.0 ...

f4exb, 2017-05-24:
I' m pretty sure OpenGL version is the problem. v 2.0 is really old. OpenGL 3.0 was released in August 2008.  I think the GLSL used in the code is from version 1.30 that appeared in OpenGL 3.0. I had to upgrade the code about a year ago to be compatible with the Qt creator version I am using to build on Windows. Some deprecated instructions were not allowed at all anymore. In any case the Windows7 based machine I am building the binaries with has OpenGL 4.3 down to 3.0 support.

DG9BFC, 2017-05-24:
so then ... give me a hint how to upgrade that
it is a win7 64 bit here
intel driver update soft did not find newer graphics driver

f4exb, 2017-05-24:
Absolutely no idea. It probably depends on your hardware.

DG9BFC, 2017-05-25:
i have checked all things in the net ... no way
there is no newer driver available ... so no higher opengl possible
seems as i have to give up :-(

f4exb, 2017-05-25:
Hello,

sorry no plans to downgrade OpenGL. I am not sure it would even build on Qt Creator

Best regards,
Edouard,

closed 2017-05-25 by f4exb

DG9BFC, 2017-05-25:
ok then no usage on laptop ... only faster machines will work

f4exb, 2017-05-25:
In any case you do need a reasonably fast machine to run near real time applications like this one. It does not really work on Raspberry Pi 3 although it would start. But it is running fine on a Pipo X8 which has a quad core Intel Z3736F running at 1.33 GHz.