f4exb/sdrangel #30: Build not finding /usr/include/cm256cc

The-Fred, 2017-06-04:
I am including the patch but can make a PR if you want. I'm not sure if this is the correct fix. It does work when I build SDRangel.

```
diff --git a/cmake/Modules/FindCM256cc.cmake b/cmake/Modules/FindCM256cc.cmake
index c48ad51a4f7e..a3f88a803b13 100644
--- a/cmake/Modules/FindCM256cc.cmake
+++ b/cmake/Modules/FindCM256cc.cmake
@@ -7,6 +7,7 @@ FIND_PATH(CM256CC_INCLUDE_DIR
     ${CMAKE_INSTALL_PREFIX}/include
     ${LIBCM256CC_INSTALL_PREFIX}/include
     PATHS
+    /usr/include/cm256cc
     /usr/local/include
     /usr/include
 )

```

f4exb, 2017-06-06:
I am not sure /usr/include/cm256cc is the correct default path when you install cm256cc. Cmake documentation says the default install prefix path is /usr/local see: https://cmake.org/cmake/help/v3.0/variable/CMAKE_INSTALL_PREFIX.html

So the includes should go to /usr/local/include that is already in the search path. Any specific path should be set with the LIBCM256CC_INSTALL_PREFIX at cmake invocation time and that path will be suffixed by /include anyway.

The-Fred, 2017-06-06:
I looked at this some more and I followed the method in the cm256cc package.   

https://github.com/f4exb/cm256cc#building-quick-setup

f4exb, 2017-06-06:
These instructions say that you should have the cm256cc installation directory as:
`cmake -Wno-dev -DCMAKE_INSTALL_PREFIX=/opt/install/cm256cc ..`

Thus on the cmake command line of sdrangel you should have among other things:
`-DCM256CC_INCLUDE_DIR=/opt/install/cm256cc/include/cm256cc -DCM256CC_LIBRARIES=/opt/install/cm256cc/lib/libcm256cc.so`

Indeed the includes are placed in a cm256cc subdirectory. I suppose I did that not to mix it up with all the other include files but then the default locations will also be:
`/usr/include/cm256cc`
`/usr/local/include/cm256cc`
`/usr/include`is useless and can be dropped

f4exb, 2017-06-11:
This is fixed in v3.5.0

closed 2017-06-11 by f4exb