f4exb/sdrangel #34: Display limeboard/chip temperature and temperature trend/stats.

The-Fred, 2017-06-06:
I noticed some code for temperature sensing but I didn't see the temperature displayed anywhere.  It might be helpful to show current temp along with history or trend stats. Maybe a gauge with a plot overlay???  Easy for me to ask for features.  :/

You may already know there is a LimeSuite API call to get the  temp in degrees C: LMS_GetChipTemperature

The part on the lime board is LM75BTP (HWSON8), datasheet:
http://www.nxp.com/documents/data_sheet/LM75B.pdf

f4exb, 2017-06-07:
Yes indeed LMS_GetChipTemperature seems to do the job. I remember now why I haven't included the temperature reading in the plugins GUI... This is a problem with multiple instances pertaining to the same device since we do not want the temperature test to be called all over the place. 

It is more complex than it seems to centralize the temperature checking among the plugin instances that pertain to the same physical device. It should probably be done using the device "buddies" and one of the "buddies" has to take the leadership to be the one that polls the temperature. Feasible but not trivial.

The-Fred, 2017-06-10:
This sounds like the correct way to do it. I will look at the buddies and see if I can figure it out and try to contribute somehow.

I noticed in the LimeSuiteGui that there are two different places that display the temperature. And they report slightly different numbers.

There is only one sensor listed in the BOM but maybe the FPGA has a temp sensor also or some other part on the board.  Also possible that they grab the temp and calculate it differently for each of the displays in LimeSuiteGui.  This would not surprise

Here I go coming up with more features...would be a nice to have thing to make the display flash or something annoying when the temps approach critical. These chips and this board is not meant to handle very high temps at all.  Different parts have different critical temps but I think I read 70C and I've seen my board report near 60C without doing a lot.

f4exb, 2017-07-03:
One way to make things not too difficult is to elect a leader among the buddies and this instance will perform the temperature reading then post a message to all its buddies. This leader feature can be made generic. 

The election takes place in the mainwindow.cpp where the source and sink instances are created or deleted.

- When adding a sink or source, if it is the only instance then it is made the leader
- When adding a sink or source, if there are any buddies then the new instance cannot be a leader
- When removing a sink or source, if it is not the leader then nothing more happens
- When removing a sink or source, if it is the leader then the first buddy found is elected the leader

f4exb, 2017-07-03:
Now in dev branch. It's a simple text display. I don't think we need something overly big and complex. Mine is stable ~50C. I have heatsinks and aluminium enclosure but no fan.

f4exb, 2017-07-07:
Implemented in v3.5.1

closed 2017-07-07 by f4exb

The-Fred, 2017-07-25:
Nice, thanks a lot!  Haven't had a chance to try it out yet.