f4exb/sdrangel #36: TX/RX (PTT) button/plugin for HackRF One.

ly2ss, 2017-07-08:
Would it be possible to implement simple RX/TX switching with single button? Kind of "Transceiver plugin", where could be also possibility to check mark "RX_freq = TX_freq" for sink and source ?
This way receive frequency would be the same.
There's a lack of FM/SSB/CW etc transceivers for HackRF and SDRangel is almost quite there.

f4exb, 2017-07-09:
HackRF is a half duplex device so I think that when you start the Tx side in the Tx tab the Rx automatically stops in the Rx tab amd vice versa. I am not sure it actually does that but it should. I am not sure the Rx and Tx frequencies can be independent either I also have to check that. For example on the LimeSDR the sample rate is the same for Rx and Tx and if you change it in any of the Rx or Tx instances it will change in all the related ones.

The "philosophy" of plugins is that they drive distinct sampling device instances materialized with the Rn and Tn tabs you see (n being a sequence number). So there cannot be such thing as a single plugin driving both Rx and Tx. There is a mechanism behind the scenes that entangles "soft' devices belonging to the same hardware device so they can communicate and possibly drive each other. Still they are represented in the software as distinct devices.

ly2ss, 2017-07-12:
You were right - once you click on T1 tab, receiver stops in R0 tab.
But for some (unknown to me) reason one must click 4 times to switch from RX to TX:
click on T1 tab
click twice on stop/play button
only on 4th click HackRF starts to send.
Switching back from TX to RX needs also 4 clicks.
This is why I was proposing some kind of "plugin" (wrong word, maybe "HackRF One transceiver" could be better) - a separate pane with one toggle PTT button.
And the same pane could have a checkmark for user to choose if he needs RX/TX frequencies to be the same or separate. Some people  need to tx/rx on the same frequency (f.e. simplex channels. This way transmitter "follows"receivers frequency). Other people  - operate on split frequencies (sats, repeaters, dx - then one should set frequencies separately in R and T tabs).
BTW:  I checked - different frequencies (2m on R0 ant 70cm on T1) work ok with HackRF One.
I do understand that all this could end with endless requests for features but it's a sad situation - such a great device but still almost no comfortable transceiver application to choose from.

f4exb, 2017-07-19:
Normally switching Rx/Tx with the "play" button should require at most 2 clicks: 1 to stop the Rx or Tx and another to start the Tx or Rx respectively. Ideally switching on Tx or Rx should stop the Rx or Tx counterpart. So the 4 clicks is a problem definitely.

I don't see a PTT button fitting very well with the plugins. However it would be possible to have an option to link and unlink Rx and Tx frequencies so they can be set independently when unlinked then can be varied accordingly when linked. Thus you can have whichever offset from Rx to Tx or no offset at all.

Once it is all set you can save the Rx and Tx configuration as presets. Possibly the Rx and Tx settings could be combined in the same preset but the practicality has to be investigated.

f4exb, 2017-07-19:
Taking the Rx to Tx example the correct sequence is stop Rx then start Tx. The nice to have is to trigger this sequence just from start Tx. From Tx to Rx the same by exchanging Rx and Tx.

f4exb, 2017-07-20:
Implemented link Tx frequency to Rx frequency option in dev branch (new button with link icon). The reverse is not implemented because of its added complexity to cancel infinite bouncing loop. So in fact Rx is "master".

ly2ss, 2017-07-22:
I am expecting that when this option activated, RX frequency from R0 tab immediately will be transferred to T1 but it is not. No matter what I do - I can't make it work: TX frequency remains unchanged. What I am possibly doing wrong? (tried to delete .config/f4exb folder too - no help)

ly2ss, 2017-07-23:
Just an idea (and only for HackRF One source/sink): to put that "PTT" button near the link frequencies button in R0 tab. I don't know if this possible but this way operation would be very easy - "point ant shoot" style and no need to switch between tabs (except maybe cw operation).
IMHO that's all the functionality needed to start easy voice AM/SSB/FM operations with HackRF One.

f4exb, 2017-07-23:
Implemented in v3.5.2

closed 2017-07-23 by f4exb

f4exb, 2017-07-23:
1. The Rx/Tx frequency link only works for the same HackRF device. Also it will work from the next frequency change after the link button has been engaged. It does not copies the Rx frequency to Tx instead it applies the Rx frequency change to Tx. This allows split and repeater operation by keeping the frequency offset between Rx and Tx. 
2. Sorry there will not be a PTT button.

watkipet, 2019-06-06:
I can control each plugin through the RESTful interface in sdrsrv even if the GUI is running, right? If so, it seems like I could have an external program do the PTT by turning off the RX plugin and turning on the TX one (assuming only one plugin for each).

Then I suppose I could connect this external program to a MIDI controller to get extra fancy.

f4exb, 2019-06-06:
`sdrangelsrv` I suppose this is what you are talking about does not have a GUI. That's the very purpose of the server variant. So I don't know what you are up to.