f4exb/sdrangel #38: Only LSB and DSB in SSB, no USB on SSB.

satfan52, 2017-07-10:
I tested using an RTL-SDR stick and the SSN de-modulator did not offer the possibility to receive in LSB ! Only DSB and USB !!! This is not convenient at all when receiving on lower bands, 40 m, 80m, 160m.
Please add this functionality.

f4exb, 2017-07-10:
Just set the filter bandwidth to a negative value

satfan52, 2017-07-11:
Thank you for pointing this out. Indeed I would have not naturally thought about it. Updating the documentation is a nice "work-around" to get going immediately on LSB. However in the interest of end-users friendliness of your very nice human interface design, I would still recommend to add to the SSB demodulator (and the corresponding add-on modulator too of course)  an LSB icon similar to the  USB one. You could frankly get rid of the DSB Icon and replace by an LSB one, unless I am mistaken it would make much more sense to most end-users of your software.

f4exb, 2017-07-11:
Sorry but I will not do that and there is no "work around" here as it works as designed. 

I don't see any reason of getting rid of the DSB/SSB toggle because this actually toggles between SSB and DSB and the demodulator does work differently in either case. If you look at the frequency scale at the bottom of the spectrum display then it becomes clear. DSB is a functionality that is maybe rarely used but I don't see any reason to remove it. It has an analog counterpart that was sometimes used as a "SSB for the poor".

The negative bandwidth limits to trigger LSB might seem strange but it does have a signification mathematically when you use complex samples and therefore allow negative frequencies. You can imagine that if you move towards negative frequencies from the center of reception that is 0 then the spectrum gets reversed. This is what LSB is about actually.

Therefore I insist that it is a matter of better documenting this behavior and not change it.

satfan52, 2017-07-12:
Well sure, up to you to see what make sense for you and for your end users ;-)

f4exb, 2017-07-22:
UI chenges now on dev branch.

f4exb, 2017-07-23:
Implemented in v3.5.2

closed 2017-07-23 by f4exb