f4exb/sdrangel #39: Broadcast FM demod: RDS is broken

f4exb, 2017-07-23:
Zero decodes when engaged

f4exb, 2017-08-02:
I have to check if the sample rate and/or RF bandwidth that should drive the channelizer downsampling have an influence

f4exb, 2017-08-11:
Well at the moment there is no documentation at all for this plugin and it is not a small job due to the many fields of RDS.

Dl4APT, 2017-10-11:
if I remember correctly RDS works fine but only in Stereo mode

f4exb, 2017-11-26:
Hello, it needs the 19 kHz pilot to be locked so most probably this is indeed tied to the stereo mode.

f4exb, 2018-02-02:
There's seem to be an issue with the sample rate. Some baseband sample rates cause it not to work. For example 384 kS/s. Maybe there's an assumption somewhere that depends on the sample rate.

f4exb, 2019-04-28:
dropped

closed 2019-04-28 by f4exb