f4exb/sdrangel #41: OSX make error:sdrangel/sdrbase/dsp/scopevis.h:4:10: fatal error: 'boost/circular_buffer.hpp' file not found

cn0xroot, 2017-07-28:
nice project.
there is something error when i build it on osx :

```
 make
[  1%] Automatic MOC for target sdrbase
Generating MOC source sdrbase_autogen/CUPUVUNIIB/moc_audiofifo.cpp
Generating MOC source sdrbase_autogen/FNPHOSZ4QI/moc_devicesinkapi.cpp
Generating MOC source sdrbase_autogen/FNPHOSZ4QI/moc_devicesourceapi.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_basebandsamplesink.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_basebandsamplesource.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_channelmarker.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_cwkeyer.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_devicesamplesink.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_devicesamplesource.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_downchannelizer.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_dspdevicesinkengine.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_dspdevicesourceengine.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_dspengine.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_samplesinkfifo.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_samplesinkfifodoublebuffered.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_samplesourcefifo.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_scopevismulti.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_threadedbasebandsamplesink.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_threadedbasebandsamplesource.cpp
Generating MOC source sdrbase_autogen/BI76252ZIR/moc_upchannelizer.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_aboutdialog.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_addpresetdialog.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_audiodialog.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_basicchannelsettingswidget.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_buttonswitch.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_channelwindow.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_clickablelabel.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_cwkeyergui.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_glscope.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_glscopegui.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_glscopemulti.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_glscopemultigui.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_glscopeng.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_glscopenggui.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_glspectrum.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_glspectrumgui.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_indicator.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_levelmeter.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_mypositiondialog.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_pluginsdialog.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_rollupwidget.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_samplingdevicecontrol.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_scale.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_valuedial.cpp
Generating MOC source sdrbase_autogen/S5ZSPE2JBJ/moc_valuedialz.cpp
Generating MOC source sdrbase_autogen/GS2YDEH7U7/moc_mainwindow.cpp
Generating MOC source sdrbase_autogen/4LCVIID3W4/moc_pluginapi.cpp
Generating MOC source sdrbase_autogen/4LCVIID3W4/moc_pluginmanager.cpp
Generating MOC source sdrbase_autogen/OLR4WOGR3K/moc_messagequeue.cpp
Generating MOC source sdrbase_autogen/OLR4WOGR3K/moc_syncmessenger.cpp
Generating MOC compilation sdrbase_autogen/moc_compilation.cpp
[  1%] Built target sdrbase_autogen
[  1%] Generating qrc_res.cpp
[  2%] Generating ui_mainwindow.h
[  2%] Generating ui_aboutdialog.h
[  2%] Generating ui_addpresetdialog.h
[  2%] Generating ui_basicchannelsettingswidget.h
[  3%] Generating ui_cwkeyergui.h
[  3%] Generating ui_glscopegui.h
[  3%] Generating ui_glscopenggui.h
[  4%] Generating ui_glscopemultigui.h
[  4%] Generating ui_glspectrumgui.h
[  4%] Generating ui_pluginsdialog.h
[  4%] Generating ui_audiodialog.h
[... 120 lines cut ...]
clang: warning: argument unused during compilation: '-fmax-errors=10' [-Wunused-command-line-argument]
In file included from /Users/cn0xroot/sdr/sdrangel/sdrbase/dsp/nullsink.cpp:1:
/Users/cn0xroot/sdr/sdrangel/sdrbase/dsp/nullsink.h:16:15: warning: 'NullSink::feed' hides overloaded virtual function [-Woverloaded-virtual]
        virtual void feed(SampleVector::const_iterator begin, SampleVector::const_iterator end, bool positiveOnly);
                     ^
/Users/cn0xroot/sdr/sdrangel/sdrbase/dsp/basebandsamplesink.h:36:15: note: hidden overloaded virtual function 'BasebandSampleSink::feed' declared here: type
      mismatch at 1st parameter ('const SampleVector::const_iterator &' (aka 'const __wrap_iter<const Sample *> &') vs 'SampleVector::const_iterator'
      (aka '__wrap_iter<const Sample *>'))
        virtual void feed(const SampleVector::const_iterator& begin, const SampleVector::const_iterator& end, bool positiveOnly) = 0;
                     ^
1 warning generated.
[ 16%] Building CXX object CMakeFiles/sdrbase.dir/sdrbase/dsp/recursivefilters.cpp.o
clang: warning: argument unused during compilation: '-fmax-errors=10' [-Wunused-command-line-argument]
[ 16%] Building CXX object CMakeFiles/sdrbase.dir/sdrbase/dsp/spectrumscopecombovis.cpp.o
clang: warning: argument unused during compilation: '-fmax-errors=10' [-Wunused-command-line-argument]
In file included from /Users/cn0xroot/sdr/sdrangel/sdrbase/dsp/spectrumscopecombovis.cpp:1:
In file included from /Users/cn0xroot/sdr/sdrangel/sdrbase/dsp/spectrumscopecombovis.h:6:
/Users/cn0xroot/sdr/sdrangel/sdrbase/dsp/scopevis.h:4:10: fatal error: 'boost/circular_buffer.hpp' file not found
#include <boost/circular_buffer.hpp>
         ^
1 error generated.
make[2]: *** [CMakeFiles/sdrbase.dir/sdrbase/dsp/spectrumscopecombovis.cpp.o] Error 1
make[1]: *** [CMakeFiles/sdrbase.dir/all] Error 2
make: *** [all] Error 2
```

f4exb, 2017-07-28:
It looks like you are missing the Boost library dependency. The Mac OS more specifically Mac OS Sierra was contributed by sygismund and I have very little knowledge about Mac.

Please make sure you read his specific readme for Mac: https://github.com/f4exb/sdrangel/blob/master/ReadmeMacOS.md

cn0xroot, 2017-07-29:
thank you all the same.

i'll try it again follow ReadmeMacOS.md

f4exb, 2017-08-14:
May be due to the fact that Boost dependency is not mandatory in the main cmake file. It should have complained at cmake stage. This was reported in issue #46 the fix is now on dev branch.

cn0xroot, 2017-08-14:
has the same result:
```
~/sdr/sdrangel/build   dev ✔   make
Scanning dependencies of target sdrbase_autogen
[  1%] Generating qrc_res.cpp
[  1%] Generating ui_mainwindow.h
[  1%] Generating ui_aboutdialog.h
[  2%] Generating ui_addpresetdialog.h
[  2%] Generating ui_basicchannelsettingswidget.h
[  2%] Generating ui_cwkeyergui.h
[  2%] Generating ui_glscopegui.h
[  3%] Generating ui_glscopenggui.h
[  3%] Generating ui_glscopemultigui.h
[  3%] Generating ui_glspectrumgui.h
[  3%] Generating ui_pluginsdialog.h
[  4%] Generating ui_audiodialog.h
[  4%] Generating ui_samplingdevicecontrol.h
[  4%] Generating ui_myposdialog.h
[  4%] Automatic MOC for target sdrbase
Generating MOC source CUPUVUNIIB/moc_audiofifo.cpp
Generating MOC source FNPHOSZ4QI/moc_devicesinkapi.cpp
Generating MOC source FNPHOSZ4QI/moc_devicesourceapi.cpp
Generating MOC source BI76252ZIR/moc_basebandsamplesink.cpp
Generating MOC source BI76252ZIR/moc_basebandsamplesource.cpp
Generating MOC source BI76252ZIR/moc_channelmarker.cpp
Generating MOC source BI76252ZIR/moc_cwkeyer.cpp
Generating MOC source BI76252ZIR/moc_devicesamplesink.cpp
Generating MOC source BI76252ZIR/moc_devicesamplesource.cpp
Generating MOC source BI76252ZIR/moc_downchannelizer.cpp
Generating MOC source BI76252ZIR/moc_dspdevicesinkengine.cpp
Generating MOC source BI76252ZIR/moc_dspdevicesourceengine.cpp
Generating MOC source BI76252ZIR/moc_dspengine.cpp
Generating MOC source BI76252ZIR/moc_samplesinkfifo.cpp
Generating MOC source BI76252ZIR/moc_samplesinkfifodoublebuffered.cpp
Generating MOC source BI76252ZIR/moc_samplesourcefifo.cpp
Generating MOC source BI76252ZIR/moc_scopevismulti.cpp
Generating MOC source BI76252ZIR/moc_threadedbasebandsamplesink.cpp
Generating MOC source BI76252ZIR/moc_threadedbasebandsamplesource.cpp
Generating MOC source BI76252ZIR/moc_upchannelizer.cpp
Generating MOC source S5ZSPE2JBJ/moc_aboutdialog.cpp
Generating MOC source S5ZSPE2JBJ/moc_addpresetdialog.cpp
Generating MOC source S5ZSPE2JBJ/moc_audiodialog.cpp
Generating MOC source S5ZSPE2JBJ/moc_basicchannelsettingswidget.cpp
Generating MOC source S5ZSPE2JBJ/moc_buttonswitch.cpp
Generating MOC source S5ZSPE2JBJ/moc_channelwindow.cpp
Generating MOC source S5ZSPE2JBJ/moc_clickablelabel.cpp
Generating MOC source S5ZSPE2JBJ/moc_cwkeyergui.cpp
Generating MOC source S5ZSPE2JBJ/moc_glscope.cpp
Generating MOC source S5ZSPE2JBJ/moc_glscopegui.cpp
Generating MOC source S5ZSPE2JBJ/moc_glscopemulti.cpp
Generating MOC source S5ZSPE2JBJ/moc_glscopemultigui.cpp
Generating MOC source S5ZSPE2JBJ/moc_glscopeng.cpp
Generating MOC source S5ZSPE2JBJ/moc_glscopenggui.cpp
Generating MOC source S5ZSPE2JBJ/moc_glspectrum.cpp
Generating MOC source S5ZSPE2JBJ/moc_glspectrumgui.cpp
Generating MOC source S5ZSPE2JBJ/moc_indicator.cpp
Generating MOC source S5ZSPE2JBJ/moc_levelmeter.cpp
Generating MOC source S5ZSPE2JBJ/moc_mypositiondialog.cpp
Generating MOC source S5ZSPE2JBJ/moc_pluginsdialog.cpp
Generating MOC source S5ZSPE2JBJ/moc_rollupwidget.cpp
Generating MOC source S5ZSPE2JBJ/moc_samplingdevicecontrol.cpp
Generating MOC source S5ZSPE2JBJ/moc_scale.cpp
Generating MOC source S5ZSPE2JBJ/moc_valuedial.cpp
Generating MOC source S5ZSPE2JBJ/moc_valuedialz.cpp
Generating MOC source GS2YDEH7U7/moc_mainwindow.cpp
Generating MOC source 4LCVIID3W4/moc_pluginapi.cpp
Generating MOC source 4LCVIID3W4/moc_pluginmanager.cpp
Generating MOC source OLR4WOGR3K/moc_messagequeue.cpp
Generating MOC source OLR4WOGR3K/moc_syncmessenger.cpp
Generating MOC compilation mocs_compilation.cpp
[... 121 lines cut ...]
/Users/cn0xroot/sdr/sdrangel/sdrbase/dsp/nullsink.h:16:15: warning: 'NullSink::feed' hides overloaded virtual function [-Woverloaded-virtual]
        virtual void feed(SampleVector::const_iterator begin, SampleVector::const_iterator end, bool positiveOnly);
                     ^
/Users/cn0xroot/sdr/sdrangel/sdrbase/dsp/basebandsamplesink.h:36:15: note: hidden overloaded virtual function 'BasebandSampleSink::feed' declared here: type
      mismatch at 1st parameter ('const SampleVector::const_iterator &' (aka 'const __wrap_iter<const Sample *> &') vs 'SampleVector::const_iterator'
      (aka '__wrap_iter<const Sample *>'))
        virtual void feed(const SampleVector::const_iterator& begin, const SampleVector::const_iterator& end, bool positiveOnly) = 0;
                     ^
1 warning generated.
[ 13%] Building CXX object CMakeFiles/sdrbase.dir/sdrbase/dsp/recursivefilters.cpp.o
clang: warning: argument unused during compilation: '-fmax-errors=10' [-Wunused-command-line-argument]
[ 14%] Building CXX object CMakeFiles/sdrbase.dir/sdrbase/dsp/spectrumscopecombovis.cpp.o
clang: warning: argument unused during compilation: '-fmax-errors=10' [-Wunused-command-line-argument]
In file included from /Users/cn0xroot/sdr/sdrangel/sdrbase/dsp/spectrumscopecombovis.cpp:1:
In file included from /Users/cn0xroot/sdr/sdrangel/sdrbase/dsp/spectrumscopecombovis.h:6:
/Users/cn0xroot/sdr/sdrangel/sdrbase/dsp/scopevis.h:4:10: fatal error: 'boost/circular_buffer.hpp' file not found
#include <boost/circular_buffer.hpp>
         ^
1 error generated.
make[2]: *** [CMakeFiles/sdrbase.dir/sdrbase/dsp/spectrumscopecombovis.cpp.o] Error 1
make[1]: *** [CMakeFiles/sdrbase.dir/all] Error 2
make: *** [all] Error 2
   ~/sdr/sdrangel/build   dev ✔ 

```

cn0xroot, 2017-08-14:
install boost with mac port:
~/sdr/sdrangel/build   dev ✔  port installed  |grep boost
  boost @1.59.0_3+no_single+no_static+python27 (active)

closed 2017-09-14 by f4exb