f4exb/sdrangel #45: FileSink crashes

wmiler, 2017-08-08:
SDRangel v3.5.4 (sdrangel64_v3.5.4.7z)
OS: win7-64

Application crashes while generating a sdriq file. Windows reports:
`Problem signature:

  Problem Event Name:	APPCRASH
  Application Name:	sdrangel.exe
  Application Version:	0.0.0.0
  Application Timestamp:	598793f4
  Fault Module Name:	sdrbase.dll
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	598792f7
  Exception Code:	40000015
  Exception Offset:	000000000009a622
  OS Version:	6.1.7601.2.1.0.256.48
  Locale ID:	1033
  Additional Information 1:	e7c0
  Additional Information 2:	e7c0bbbc8e445e5e37d96a5ebdf48134
  Additional Information 3:	d7c7
  Additional Information 4:	d7c7b34d85c05483bbeb88be436499b8`

f4exb, 2017-08-08:
Unfortunately this doesn't tell much. I could try to reproduce if you give me the exact configuration of plugins and their settings.

wmiler, 2017-08-09:
This doesn't happen on my win7-64 development system. I'll have to play with the laptop and see what is going on.

closed 2017-09-14 by f4exb