f4exb/sdrangel #47: SDRDaemon plugin is not in main build anymore

F5OEO, 2017-08-11:
Fresh install and see in the documentation, SDRDaemon plugin is missing from build..or I miss something. A commity is about to make some strike with a heavy slogan "Get Daemon back !". Feedback to solve the issue is more than welcome !

f4exb, 2017-08-11:
Hello my friend!
SDRdaemon plugin has specific dependencies that if not met will prevent the build. These are:
  - libnanomsg (get it from distribution)
  - cm256cc (get it from https://github.com/f4exb/cm256cc)
Let us know if this solves the problem

F5OEO, 2017-08-11:
Linux users are finally lazy ones...Even documented, general usage seems to be copy/paste of commandlines. In that concept, there are some missing parts.

f4exb, 2017-08-12:
So is it working now?

F5OEO, 2017-08-12:
"Yes captain" : Compiling cm256cc does the trick (RTFM)

closed 2017-08-12 by F5OEO