f4exb/sdrangel #48: Windows eradication

f4exb, 2017-08-11:
Discontinue Windows support and remove all related files.

F5OEO, 2017-08-11:
General strike ... why not ?

f4exb, 2017-08-11:
In the end I cannot spend time on problems that are not with my software but with the O/S.

f4exb, 2017-08-12:
This is also me getting angry at totally loosing my laptop installation due to a Win10 update walking over grub although it was not on the same volume.

Anyway at least Windows support is under close observation. In fact I was trying to run the latest version of SDRangel on Win10 because some problems were reported. Indeed a lot of plugins are not there without a valid reason since on Win7 everything worsk fine.

Sure I don't want to spend time on this kind of issue and move on to more feature and real bug fixes.

f4exb, 2017-08-13:
After a full reinstall on the Win10 Vbox machine things look a lot better. I can even run SDRangel inside the Win10 Vbox machine which used to break at OpenGL init time.

f4exb, 2017-08-13:
The same build works on my Win7 box so I think it is going to be the deal.

closed 2017-08-13 by f4exb