f4exb/sdrangel #49: Macros and actions

f4exb, 2017-08-12:
It was recently suggested on the MyriadRF forum that there could be a keyboard shortcut to switchover Rx/Tx in a transceiver type of arrangement. Even earlier someone suggested a "PTT button". 

This behavior may not be always desirable for full duplex devices and a macro driving a sequence of predefined actions (here stop device on one tab and start on another and vice versa) would handle this nicely. Moreover it is flexible enough to associate Rx and Tx from different physical devices.

In the actions possible there should be the invocation of a shell command in order for example to drive some GPIO pins that in turn drive PA/LNA switchover. Again it is flexible enough so you can put everything you want even lit a big "on air" sign!

f4exb, 2017-08-12:
This could be rally cool not only for the "PTT" thing. One could drive a complete setup like this. I often open 3 Rx tabs with different receiver to listen to different bands with different presets. So among the actions there could be:

  - Add source device
  - Add sink device
  - Load a preset in the last open (created) tab or by tab number

This way you can load a complete configuration and then since you know where your tabs are you can safely activate the "PTT" macro.

ly2ss, 2017-08-12:
Actually it was me suggesting PTT feature and my suggestion was to implement it for single siplex hackrf device.
Now seeing to this post I thought about single keyboard key assigned to some rx/tx macro? 
I am a bit puzzled why one would need to jump to TX tab to see his transmitted CW/FM/whatever waveform/fft ?
I still believe I can not fully grasp philosophy/flexibility of your project and hence my feature requests might appear to you a bit foolish and somehow limiting flexibility of your creature.

73 Zilvis LY2SS

f4exb, 2017-08-14:
In fact I am the main developer and also the "product owner" which in strict agility should be handled by two different parties but in a team of one the rules become a bit special :smile: Having worked on it for more than 2 years and using it every day I have some ideas on what it should and should not be.

I made the decision already some time ago to have Rx and Tx in separate tabs even if the relate to the same device. This is what made the most sens to me by just reversing the Rx flow for the Tx. This is why you need to move the focus to the Tx tab to start the Tx. The nice thing abut this macro idea is that the change of focus could be made automatic and tied to a key. Qt has probably the provision to handle press and release events separately which allows for a real PTT.

SDRangel is certainly different of the other SDR software that imitate a real radio. It was the idea of the original authors (of SDRangelove) when I took over the project and I had special care to keep this spirit.

f4exb, 2017-08-22:
#53 is a prerequisite

brendanthebig, 2017-11-09:
If implementing PTT and sequencer control using GPIO, would it be possible to add in TX inhibit until sequencer reports it is in correct state ?

mntnmn, 2017-11-10:
Whatever way you end up doing this, it would be outstanding to present a CAT interface on a virtual serial port. Many other programs could then interface with SDRangel as a rig. This may get WSJT, fldigi, and the like to work more easily with SDRangel.

f4exb, 2017-11-21:
The CAT can be handled in another project/software or even just a script can do that. This program will support a web REST API **exclusively**.

More thoughts on the Macro feature:

  - The list could be presented in a docked widget like the presets
  - It will certainly be an external call preferably targeting a Python script but not necessarily exclusively (see: https://stackoverflow.com/questions/15127047/qt-calling-external-python-script). It will need Swagger client definitions and Python is convenient for this.
  - It can be run from selection or linked to a keyboard action (key press or release or just key hit), 
  - Only the addressing details (host address and port) are given as parameter. All application data should be accessible via the web API.
  - A built in editor supporting Python syntax highlighting could be provided as a convenience (using QPlainTextEdit and QSyntaxHighlighter see: https://talktomachines.blogspot.fr/2014/01/python-syntax-highlighting-for-qtextedit.html)

f4exb, 2018-01-03:
The editor may be left as a future nice to have. Confirming this will wrap a QProcess invocation to an external program. 

Optional parameters derived from the UI current state:

- Rx or Tx focused device set indication
- Focused device set index
- Focused channel index

f4exb, 2018-01-07:
Done in v 3.10.0

closed 2018-01-07 by f4exb