f4exb/sdrangel #51: Controlling multiple LimeSDR sources

Analias, 2017-08-19:
Getting odd behavior when using multiple LimeSDR sources. When I start up the app, it defaults to one of my LimeSDR devices as R0. I then add a second LimeSDR source as R1. I'm being careful to not use the same LimeSDR device by verifying the serial numbers differ on the sampling device dropdown selection.

If I then try to start either LimeSDR source the app becomes non-responsive and I have to kill it. If I select a preset for either LimeSDR source (R0 or R1) it appears to set both of them. If I try to change the frequency for either source nothing happens.

If I start the defaulted LimeSDR source before adding the second one it works. If I try to change the frequency for either source, it appears to work. If I select a preset, it effects both LimeSDR sources.

I have a third LimeSDR I could try plugging in and see if the problem changes with three LimeSDR devices but expect it to behave oddly.

--------------------

 While typing up the above and verifying how to reproduce the behavior I must have tried saving a preset. Now both LimeSDR devices default to 0.000.000 frequency. If I try to change the main frequency either LimeSDR source, the NCO frequency zeros and I cannot change anything. If I try to start either LimeSDR source with the frequency at 0.000.000 I get a pop-up info window saying "Could not start sample source".

I also have an attached RTL-SDR. If I change either R0 or R1 to use the RTL-SDR it works. Switching back to the LimeSDR exhibits to the same behavior.

-Freeman, N5FPP

f4exb, 2017-08-22:
Unfortunately I only have one LimeSDR to play with but I can possibly reproduce the case with other devices (I have several HackRFs) since the process is generic. I suspect an issue when the first Lime is the default device selected in R0 at startup because in this case things are handled differently. Normally changes in one device in one tab may make changes in other tabs if they pertain to the same device identified by its type and serial number. I'm pretty sure it works for R1, R2, ... but may be there's bug with R0.

f4exb, 2017-08-22:
No problem whatsoever with two HackRFs. I think it is a problem of driver particularly if you run on Windows. I'm afraid the flows for the two Limes gets messed up.

The zero display as default frequency is typical of the Lime not being responsive in which case I usually unplug and plug again.

Analias, 2017-08-22:
I'll have to test that hypothesis. I will try running two instances of gqrx and see if I get the same behavior. I will let you know what I find.

Analias, 2017-08-23:
After disconnecting and reconnecting both my LimeSDRs I was able to run two instances of gqrx using both LimeSDRs without any problem. I tried reproducing the problem in sdrangel and both worked together without any interaction. 

I will try later today and see if I can get sdrangel to misbehave.

f4exb, 2017-09-23:
In any case you should try with v3.7.2 (or future revision from this one on) that is compiled with LimeSuite 17.09. This version of LimeSuite appears to have important bug fixes.

closed 2017-10-19 by f4exb