f4exb/sdrangel #52: Multiple sources can be assigned the same device

Analias, 2017-08-19:
While testing a previous bug I submitted. I noticed it is possible to add multiple source devices (R0, R1, etc.) and assign the same source device to all of them. This seems counter intuitive and should probably not be allowed - or at the very least a warning message saying the same device has been assigned to multiple sources.

If restricting assigning the same device to multiple devices is desired. I would recommend marking devices in the drop-down selection for other sampling devices as disabled depending if they have been previously assigned.

-Freeman, N5FPP

f4exb, 2017-08-22:
I agree this is not foolproof assumption is that the user doesn't try to do odd things.

f4exb, 2017-10-27:
The management of devices should be reworked in a better way. There are problems reported through other means than this board that multiple Rx or Tx (for now only LimeSDR) are not working properly. For example one cannot select the second Rx if the first is not already open. 

This boils down to having a better enumeration of devices that can be more dynamic and which changes can be reported to appropriate listeners so that the actually available devices lists are updated.

texasyojimbo, 2017-10-31:
This actually seems more relevant to my comment about channel confusion on Issue #86

f4exb, 2017-10-31:
More elaboration on this subject:
  - Enumeration should be done and maintained (as plugins take and leave devices) in a single place as a singleton or attached to a singleton of the application
  - The drop down list of devices and the checkbox need to be replaced by a dialog that will be filled at the dialog open time with the current list of available devices. The "OK" button of the dialog will validate the change. Thus this is made dynamic and also is probably more intuitive.

f4exb, 2017-11-03:
Implemented in v3.8.0

closed 2017-11-03 by f4exb