f4exb/sdrangel #54: Offset tuning to simplify use with up(down)converters?

kovinesinsila, 2017-09-05:
Hello,

Are there any plans to implement this feature?
Thanks!

f4exb, 2017-09-06:
I thought of it for the DX patrol which basically is a RTLSDR and a converter for HF bands combined. This could be useful in any case for microwave transverters operation for example. However I wouldn't like the setting to add more stuff to the plugin GUI so it could be set with a popup menu a bit like the channel marker settings for channel plugins.

f4exb, 2017-09-23:
This is very complex and has not a high priority so it will not be implemented any time soon

f4exb, 2017-09-24:
Found a way to make it less difficult by exploiting the part that already does frequency shifting to move DC Away (supradyne/infradyne modes)

f4exb, 2017-09-24:
Functionality available on dev branch for RTLSDR

f4exb, 2017-09-25:
Added Airspy, Funcube Pro and Pro+, PlutoSDR Rx and Tx

f4exb, 2017-10-04:
Implemented in v3.7.3 for Airspy, Funcube Pro and Pro+, PlutoSDR Rx and Tx, and RTLSDR

closed 2017-10-04 by f4exb