f4exb/sdrangel #56: PlutoSDR Tx support

f4exb, 2017-09-11:
Opened to complete PlutoSDR support without delaying the PlutoSDR Rx support.

f4exb, 2017-09-11:
This will also include refactoring to better split between model and view. For LimeSDR for example when a change occurs on one side (Rx, Tx) the necessary changes on the other side are propagated through the GUI which is a very bad design in view of the future headless version. A similar mechanism is necessary for PlutoSDR Rx/Tx propagation.

rgetz, 2017-09-18:
I have been playing with PlutoSDR Rx, and this seems to work pretty well (Thanks for the work).

You are correct - while the LOs are completely independent - the ADC and the DAC interface clock rates are linked. Which isn't super intuitive to someone not familiar with the underlying chips.

If you have questions - please let me know.

f4exb, 2017-09-21:
Implemented in v3.7.1

closed 2017-09-21 by f4exb