f4exb/sdrangel #58: Feature request

mntnmn, 2017-09-14:
Is there a way to work in something like the modems here? Tx and Rx?

https://github.com/atmatthewat

f4exb, 2017-09-16:
I suppose you're talking about the xxx-encode-decode repositories. Not much is said about these protocols and what kind of radio interface is expected. They appear to be more or less well known however:

MDC-1200: https://en.wikipedia.org/wiki/MDC-1200
This is AFSK most probably over FM but could be anything supporting audio modulation.
It could be fully integrated within SDRangel but for this kind of configuration I would suggest sending the output of the demod via USB to an external decoder based on the mentioned library. The audio demod part is available right away for FM, AM, SSB.

FleetSink: less information as this is Kenwood proprietary. It seems to be AFSK as well so the same as for MDC-1200 applies

GE-Star: should also be AFSK so again the same thing...

I found a short video demonstrating they are all AFSK: https://www.youtube.com/watch?v=mZ3dNkoKjVk

For the moment I do not foresee implementing all possible protocols as long as the UDP connection provides a solution.

A ticket was opened earlier in which I put details on how to make APRS work which is also AFSK: https://github.com/f4exb/sdrangel/issues/25

Since version 3.6.0 an UDP sink plugin exists to support transmission so the other way round to transmit signals should be possible.

f4exb, 2017-09-23:
As said please use the UDP source plugin

closed 2017-09-23 by f4exb