f4exb/sdrangel #59: Build fail under Kali 2 rolling

friedri1970, 2017-09-16:
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp: In member function ‘bool SDRPlayInput::openDevice()’:
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:79:37: error: ‘MIRISDR_HW_SDRPLAY’ was not declared in this scope
     if ((res = mirisdr_open(&m_dev, MIRISDR_HW_SDRPLAY, m_devNumber)) < 0)
                                     ^~~~~~~~~~~~~~~~~~
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:79:37: note: suggested alternative: ‘MIRISDR_API’
     if ((res = mirisdr_open(&m_dev, MIRISDR_HW_SDRPLAY, m_devNumber)) < 0)
                                     ^~~~~~~~~~~~~~~~~~
                                     MIRISDR_API
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp: In member function ‘virtual bool SDRPlayInput::start()’:
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:119:13: error: ‘mirisdr_set_sample_format’ was not declared in this scope
  if ((res = mirisdr_set_sample_format(m_dev, s12FormatString))) // sample format S12
             ^~~~~~~~~~~~~~~~~~~~~~~~~
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:119:13: note: suggested alternative: ‘mirisdr_set_sample_rate’
  if ((res = mirisdr_set_sample_format(m_dev, s12FormatString))) // sample format S12
             ^~~~~~~~~~~~~~~~~~~~~~~~~
             mirisdr_set_sample_rate
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:137:13: error: ‘mirisdr_set_transfer’ was not declared in this scope
  if ((res = mirisdr_set_transfer(m_dev, bulkFormatString)) < 0)
             ^~~~~~~~~~~~~~~~~~~~
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:137:13: note: suggested alternative: ‘mirisdr_reset_buffer’
  if ((res = mirisdr_set_transfer(m_dev, bulkFormatString)) < 0)
             ^~~~~~~~~~~~~~~~~~~~
             mirisdr_reset_buffer
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp: In member function ‘bool SDRPlayInput::applySettings(const SDRPlaySettings&, bool, bool)’:
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:303:35: error: ‘mirisdr_get_mixbuffer_gain’ was not declared in this scope
                         lnaGain = mirisdr_get_mixbuffer_gain(m_dev);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:303:35: note: suggested alternative: ‘mirisdr_get_tuner_gain’
                         lnaGain = mirisdr_get_mixbuffer_gain(m_dev);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                   mirisdr_get_tuner_gain
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:307:35: error: ‘mirisdr_get_lna_gain’ was not declared in this scope
                         lnaGain = mirisdr_get_lna_gain(m_dev);
                                   ^~~~~~~~~~~~~~~~~~~~
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:307:35: note: suggested alternative: ‘mirisdr_get_tuner_gain’
                         lnaGain = mirisdr_get_lna_gain(m_dev);
                                   ^~~~~~~~~~~~~~~~~~~~
                                   mirisdr_get_tuner_gain
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:312:29: error: ‘mirisdr_get_mixer_gain’ was not declared in this scope
                             mirisdr_get_mixer_gain(m_dev),
                             ^~~~~~~~~~~~~~~~~~~~~~
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:312:29: note: suggested alternative: ‘mirisdr_get_tuner_gain’
                             mirisdr_get_mixer_gain(m_dev),
                             ^~~~~~~~~~~~~~~~~~~~~~
                             mirisdr_get_tuner_gain
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:313:29: error: ‘mirisdr_get_baseband_gain’ was not declared in this scope
                             mirisdr_get_baseband_gain(m_dev),
                             ^~~~~~~~~~~~~~~~~~~~~~~~~
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:313:29: note: suggested alternative: ‘mirisdr_get_tuner_gain’
                             mirisdr_get_baseband_gain(m_dev),
                             ^~~~~~~~~~~~~~~~~~~~~~~~~
                             mirisdr_get_tuner_gain
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:332:29: error: ‘mirisdr_set_mixbuffer_gain’ was not declared in this scope
                     int r = mirisdr_set_mixbuffer_gain(m_dev, settings.m_lnaOn ? 0 : 1); // mirisdr_set_mixbuffer_gain takes gain reduction
                             ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:332:29: note: suggested alternative: ‘mirisdr_set_tuner_gain’
                     int r = mirisdr_set_mixbuffer_gain(m_dev, settings.m_lnaOn ? 0 : 1); // mirisdr_set_mixbuffer_gain takes gain reduction
                             ^~~~~~~~~~~~~~~~~~~~~~~~~~
                             mirisdr_set_tuner_gain
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:346:29: error: ‘mirisdr_set_lna_gain’ was not declared in this scope
                     int r = mirisdr_set_lna_gain(m_dev, settings.m_lnaOn ? 0 : 1); // mirisdr_set_lna_gain takes gain reduction
                             ^~~~~~~~~~~~~~~~~~~~
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:346:29: note: suggested alternative: ‘mirisdr_set_tuner_gain’
                     int r = mirisdr_set_lna_gain(m_dev, settings.m_lnaOn ? 0 : 1); // mirisdr_set_lna_gain takes gain reduction
                             ^~~~~~~~~~~~~~~~~~~~
                             mirisdr_set_tuner_gain
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:365:25: error: ‘mirisdr_set_mixer_gain’ was not declared in this scope
                 int r = mirisdr_set_mixer_gain(m_dev, settings.m_mixerAmpOn ? 0 : 1); // mirisdr_set_lna_gain takes gain reduction
                         ^~~~~~~~~~~~~~~~~~~~~~
/root/software/plutosdr/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.cpp:365:25: note: suggested alternative: ‘mirisdr_set_tuner_gain’
                 int r = mirisdr_set_mixer_gain(m_dev, settings.m_mixerAmpOn ? 0 : 1); // mirisdr_set_lna_gain takes gain reduction
                         ^~~~~~~~~~~~~~~~~~~~~~
                         mirisdr_set_tuner_gain
compilation terminated due to -fmax-errors=10.
plugins/samplesource/sdrplay/CMakeFiles/inputsdrplay.dir/build.make:90: die Regel für Ziel „plugins/samplesource/sdrplay/CMakeFiles/inputsdrplay.dir/sdrplayinput.cpp.o“ scheiterte
make[2]: *** [plugins/samplesource/sdrplay/CMakeFiles/inputsdrplay.dir/sdrplayinput.cpp.o] Fehler 1

f4exb, 2017-09-16:
It looks like you do not have libmirisdr installed or the paths configure correctly. Make sure you read the readme.md first:  https://github.com/f4exb/sdrangel#sdrplay-rsp1

closed 2017-09-23 by f4exb

protektwar, 2022-03-10:
I have libmirisdr installed, but still I'm getting these errors... the link #sdrplay-rsp1 is not available anymore... 
How can I solve this error?

f4exb, 2022-03-12:
https://github.com/f4exb/sdrangel/wiki/Sample-source-plugins-(Rx-devices)#sdrplay-rsp1