f4exb/sdrangel #61: How do you configure SDRAngel to know the correct uri for ADALM-PLUTO

G4DCP, 2017-09-17:
My Adalm-pluto is network connected to my PC.  For example iio_info -n pluto.local will show it's status.

How do I tell SDRAngel the information -n pluto.local ?

Peter G4DCP

f4exb, 2017-09-17:
Hello,
it works only with UDP connected devices.
Best regards.

G4DCP, 2017-09-17:
As the difference  between the two is a single parameter to a single call can you let me know if you could make this a feature request.

rgetz, 2017-09-18:
As it would be nice to connect to Pluto via WiFi and Ethernet USB dongles. However, since it would need a IP field, to type in the IP number, I'm not sure how to suggest how it's actually done without messing up the GUI too much. A callback in libiio is on the todo list, based on zeroconf/Avahi - but its not complete yet ( analogdevicesinc/libiio#121 ) . If that is the way to keep things clean (which it might be - please comment there, and we can have a look sooner than later).

Thanks
-Robin

f4exb, 2017-09-18:
The code is here: https://github.com/f4exb/sdrangel/blob/master/devices/plutosdr/deviceplutosdrscan.cpp As you can see it uses the scan method and stores the correspondence between the serial number and the URI then the device is managed with the serial number and the URI is found with correspondence map. If the network devices can be scanned then they would be treated the same way.

G4DCP, 2017-09-18:
Might the answer be to use the pluto's zeroconfig feature. Then one could just look for an ip connected pluto by trying to resolve pluto.local, if it resolves then you have the ip address for the local pluto?

Peter

rgetz, 2017-09-18:
Yes, but it can take a long time(TM) for avahia to return - it's certainly not as immediate as the USB scan feature.

nsa666666, 2017-09-19:
You could scan for the config file on the pluto mounted disk. It would be a dirty hack tough.

closed 2017-11-05 by f4exb