f4exb/sdrangel #64: SDRangel is crashing (Windows7)

tom-bavaria, 2017-09-23:
Starting SDRangel   (sdrangel64_v3.7.2.7z) in Windows 7 (Pro., 64 bit)  I'm getting this crash infos:
(sorry for german version) 
`Protokollname: Application
Quelle:        Application Error
Datum:         23.09.2017 16:39:15
Ereignis-ID:   1000
Aufgabenkategorie:(100)
Ebene:         Fehler
Schlüsselwörter:Klassisch
Benutzer:      Nicht zutreffend
Computer:      Thomas-THINK
Beschreibung:
Name der fehlerhaften Anwendung: sdrangel.exe, Version: 0.0.0.0, Zeitstempel: 0x59c5f262
Name des fehlerhaften Moduls: ig4icd64.dll, Version: 8.15.10.2401, Zeitstempel: 0x4dd7fbd4
Ausnahmecode: 0xc000041d
Fehleroffset: 0x00000000000646ae
ID des fehlerhaften Prozesses: 0xe2c`

The modul "ig4icd64.dll" is part of an Intel graphic driver, there is no update available. 
(checked with "Intel Driver & Support Assistant Tray")

Do you have any suggestions?

f4exb, 2017-09-23:
Sorry your hardware graphics is not supported. It is too old probably.

tom-bavaria, 2017-09-23:
It's a Lenovo Thinkpad T410 from 2010/2011 (with optimus grahpics). Do I need a newer notebook?

f4exb, 2017-09-24:
Just a problem of switching to the right graphics card then. I suppose NVIDIA has supported OpenGL 3.0+ since a long time ago.

closed 2017-09-24 by f4exb