f4exb/sdrangel #65: Sdrangel hangs on exit with FUTEX_WAIT_PRIVATE.

kovinesinsila, 2017-09-26:
Hello,
The process doesn't terminate on window close and hangs on FUTEX_WAIT_PRIVATE.
stderr:
```
$ sdrangel                        
QMetaObject::connectSlotsByName: No matching signal for on_sampleSource_confirmClicked(bool)
QMetaObject::connectSlotsByName: No matching signal for on_sampleSink_confirmClicked(bool)
QMetaObject::connectSlotsByName: No matching signal for on_channel_addClicked(bool)
PluginManager::loadPlugins:  Failed to extract plugin meta data from '/opt/sdrangel/lib/libbladerfdevice.so'
PluginManager::loadPlugins:  Failed to extract plugin meta data from '/opt/sdrangel/lib/libfcdhid.so'
PluginManager::loadPlugins:  Failed to extract plugin meta data from '/opt/sdrangel/lib/libfcdlib.so'
PluginManager::loadPlugins:  Failed to extract plugin meta data from '/opt/sdrangel/lib/libhackrfdevice.so'
PluginManager::loadPlugins:  Failed to extract plugin meta data from '/opt/sdrangel/lib/libhttpserver.so'
PluginManager::loadPlugins:  Failed to extract plugin meta data from '/opt/sdrangel/lib/libsdrbase.so'
PluginManager::loadPlugins: loaded plugin libchanalyzer.so
PluginManager::loadPlugins: loaded plugin libchanalyzerng.so
PluginManager::loadPlugins: loaded plugin libdemodam.so
PluginManager::loadPlugins: loaded plugin libdemodatv.so
PluginManager::loadPlugins: loaded plugin libdemodbfm.so
PluginManager::loadPlugins: loaded plugin libdemoddsd.so
PluginManager::loadPlugins: loaded plugin libdemodlora.so
PluginManager::loadPlugins: loaded plugin libdemodnfm.so
PluginManager::loadPlugins: loaded plugin libdemodssb.so
PluginManager::loadPlugins: loaded plugin libdemodtcpsrc.so
PluginManager::loadPlugins: loaded plugin libdemodudpsrc.so
PluginManager::loadPlugins: loaded plugin libdemodwfm.so
PluginManager::loadPlugins: loaded plugin libmodam.so
PluginManager::loadPlugins: loaded plugin libmodnfm.so
PluginManager::loadPlugins: loaded plugin libmodssb.so
PluginManager::loadPlugins: loaded plugin libmodudpsink.so
PluginManager::loadPlugins: loaded plugin libmodwfm.so
PluginManager::loadPlugins: loaded plugin liboutputbladerf.so
PluginManager::loadPlugins: loaded plugin liboutputfilesink.so
PluginManager::loadPlugins: loaded plugin liboutputhackrf.so
PluginManager::loadPlugins: loaded plugin liboutputsdrdaemonsink.so
PluginManager::loadPlugins: loaded plugin libinputairspy.so
PluginManager::loadPlugins: loaded plugin libinputbladerf.so
PluginManager::loadPlugins: loaded plugin libinputfcdpro.so
PluginManager::loadPlugins: loaded plugin libinputfcdproplus.so
PluginManager::loadPlugins: loaded plugin libinputfilesource.so
PluginManager::loadPlugins: loaded plugin libinputhackrf.so
PluginManager::loadPlugins: loaded plugin libinputrtlsdr.so
PluginManager::loadPlugins: loaded plugin libinputsdrdaemonsource.so
PluginManager::loadPlugins: loaded plugin libinputsdrplay.so
Found Rafael Micro R820T tuner
RTLSDRInput::openDevice: open: Realtek RTL2838UHIDIR, SN: 00000054
QMetaObject::connectSlotsByName: No matching signal for on_sampleRate_currentIndexChanged(int)
Exact sample rate is: 230000,000533 Hz
could not set LO ppm correction: 54
Exact sample rate is: 230000,000533 Hz
AudioOutput::start: using default device alsa_output.pci-0000_00_1b.0.analog-stereo
AudioOutput::start: audio format OK
SDRangel quit.
^C
$
```
backtrace:
```
#0  0x00007fddcc24738d in pthread_cond_wait@@GLIBC_2.3.2 () at /usr/lib/libpthread.so.0
#1  0x00007fddce0d65cc in QWaitCondition::wait(QMutex*, unsigned long) () at /usr/lib/libQt5Core.so.5
#2  0x00007fddce0d4d07 in QThread::wait(unsigned long) () at /usr/lib/libQt5Core.so.5
#3  0x00007fddcf76afd8 in DSPDeviceSourceEngine::~DSPDeviceSourceEngine() (this=0x560ff2e49fe0, __in_chrg=<optimized out>) at /home/me/builds/sdrangel-git/src/sdrangel/sdrbase/dsp/dspdevicesourceengine.cpp:54
#4  0x00007fddcf76b0c9 in DSPDeviceSourceEngine::~DSPDeviceSourceEngine() (this=0x560ff2e49fe0, __in_chrg=<optimized out>) at /home/me/builds/sdrangel-git/src/sdrangel/sdrbase/dsp/dspdevicesourceengine.cpp:55
#5  0x00007fddcf768e82 in DSPEngine::~DSPEngine() (this=0x7fddcfa7d700 <(anonymous namespace)::Q_QGS_dspEngine::innerFunction()::holder>, __in_chrg=<optimized out>)
    at /home/me/builds/sdrangel-git/src/sdrangel/sdrbase/dsp/dspengine.cpp:46
#6  0x00007fddcf768f09 in (anonymous namespace)::Q_QGS_dspEngine::Holder::~Holder() (this=<optimized out>, __in_chrg=<optimized out>) at /home/me/builds/sdrangel-git/src/sdrangel/sdrbase/dsp/dspengine.cpp:51
#7  0x00007fddcd70b488 in __run_exit_handlers () at /usr/lib/libc.so.6
#8  0x00007fddcd70b4da in  () at /usr/lib/libc.so.6
#9  0x00007fddcd6f4f71 in __libc_start_main () at /usr/lib/libc.so.6
#10 0x0000560ff0b8249a in _start ()
```
strace:
[sdrangel_strace.zip](https://github.com/f4exb/sdrangel/files/1333539/sdrangel_strace.zip)
Thank you.

f4exb, 2017-09-27:
It's been a long time since I haven't audited this part of the code. So having a new look I found that the chain of actions at destruction time of DSPDeviceSourceEngine is somewhat strange:

DSPDeviceSourceEngine::~DSPDeviceSourceEngine()
{
	wait();
}

So far so good but I suppose the stop method is called some time before:

void DSPDeviceSourceEngine::stop()
{
	qDebug() << "DSPDeviceSourceEngine::stop";
	DSPExit cmd;
	m_syncMessenger.sendWait(cmd);
}

Why not just QThread::stop()? Anyway the SyncMessenger does this:

int SyncMessenger::sendWait(Message& message, unsigned long msPollTime)
{
    m_message = &message;
	m_mutex.lock();
	m_complete.testAndSetAcquire(0, 1);

	emit messageSent();

	while (!m_complete.testAndSetAcquire(0, 1))
	{
		m_waitCondition.wait(&m_mutex, msPollTime);
	}

	m_complete = 0;
	int result = m_result;
	m_mutex.unlock();

	return result;
}

Here you get the mutex. The emit is catched by the handleSynchronousMessages method of  DSPDeviceSourceEngine  and executes this bit of code:

	if (DSPExit::match(*message))
	{
		gotoIdle();
		m_state = StNotStarted;
		exit();
	}

not sure "exit" is really what we want here... ?

f4exb, 2017-09-30:
I have set up a 17.04 machine in VirtualBox recompiled everything there but I still can't reproduce the problem.

f4exb, 2017-10-03:
Maybe executing the code directly in the stop method without queuing on the messenger would circumvent the transient mutex problem:

    void DSPDeviceSourceEngine::stop()
    {
        qDebug() << "DSPDeviceSourceEngine::stop";
        gotoIdle();
        m_state = StNotStarted;
        QThread::exit();
//	DSPExit cmd;
//	m_syncMessenger.sendWait(cmd);
    }

f4exb, 2017-10-04:
Possibly fixed in v3.7.3

closed 2017-10-04 by f4exb