f4exb/sdrangel #66: libnanomsg0 does not exist in ubuntu 17.04

simonvik, 2017-09-27:
`libnanomsg0` is defined as dependency in the debian control file but the replacement would be 
`libnanomsg4` I guess that something like `libnanomsg0 | libnanomsg4` might work.

f4exb, 2017-09-27:
Yes 17.04 (Zesty) has libnanomsg4 not libnanomsg0. Checking if your suggestion would work for both 16.04 and 17.04

f4exb, 2017-09-27:
Another option although not very clean is to put libnanomsg-dev as a dependency so the right version is picked up as a dependency of libnanomsg-dev.

simonvik, 2017-09-27:
Correct, the rest of the build-deps should also be added.. ubuntus build defines:
```
               cmake,
               debhelper (>= 9),
               gnuradio (>> 3.7.9),
               gnuradio-dev (>> 3.7.9),
               gr-osmosdr,
               libboost-dev,
               libboost-system-dev,
               libfftw3-dev,
               liblog4cpp5-dev,
               libosmosdr-dev,
               libqt5opengl5-dev,
               librtlsdr-dev,
               libusb-1.0-0-dev [linux-any],
               libusb2-dev [kfreebsd-any],
               pkg-config,
               qt5-qmake,
               qtbase5-dev,
               qtbase5-dev-tools,
               qtchooser,
               qtmultimedia5-dev
```

f4exb, 2017-09-30:
No... these are needed to build the package not to run it. Moreover there are errors: gnuradio and gr-osmosdr are not needed.

The "|" syntax does work. This is how it should look like:
    
    apt-cache depends sdrangel
    sdrangel
      Depends: libc6
      Depends: libasound2
      Depends: libfftw3-single3
      Depends: libgcc1
      Depends: libgl1-mesa-glx
     |Depends: <libnanomsg0>
      Depends: libnanomsg4
      Depends: libqt5core5a
      Depends: libqt5gui5
      Depends: libqt5multimedia5
      Depends: libqt5network5
      Depends: libqt5opengl5
      Depends: libqt5widgets5
      Depends: libqt5multimedia5-plugins
      Depends: libstdc++6
      Depends: libusb-1.0-0
      Depends: libopencv-dev
      Depends: libsqlite3-dev
      Depends: pulseaudio
        pulseaudio:i386
      Depends: libxml2
      Depends: libopencv-core2.4v5
      Depends: libopencv-highgui2.4-deb0
      Depends: libopencv-imgproc2.4v5
     |Depends: libqt5gui5
      Depends: libqt5gui5-gles
     |Depends: libqt5opengl5
      Depends: libqt5opengl5-gles
      Depends: libsqlite3-0

f4exb, 2017-10-04:
Fixed in v3.7.3

closed 2017-10-04 by f4exb