f4exb/sdrangel #67: Switching between Source (receive) and Sink (transmit) - sink does not respond

martywittrock, 2017-09-27:
When I load a sink device for transmit and I switch off the receive (source) and switch on the transmit (sink) stream, the transmit will work fine for as long as I choose to operate it, but when I switch off the transmit device and then try to switch on the receive, the receive does not start up. It seems to hang and I have to restart the application completely (leave SDRAngel and then restart SDRAngel and then reload all my presets, etc). Not sure if this matters or not, but this is all being worked on the HF bands (generally testing it on 40m).

Can you check this? This is on the Win64 build and it's been working on receive wonderfully on my LattePanda board with the 7" touch screen and the microPC that I intend to embed to make the wideband transceiver using the LimeSDR. There has been A LOT of interest in my work with SDRAngel running on the LattePanda board on Win10 and many hams appear to be putting this configuration together for their own transceivers - it's a great combination for a VERY portable rig.

Keep us advised on the issues, Edouard - 73

Marty, KN0CK

f4exb, 2017-09-29:
I suppose you're dealing with the LimeSDR.

martywittrock, 2017-09-29:
Edouard,

But it's the interface itself and how it handles transmit and receive switching that's at play here. What is the proper procedure to leave receive mode and enter transmit using SDRAngel with any radio (not just LimeSDR)? Does this occur in the other radios (Ettus, etc)? If this is LimeSDR related, then what bottleneck is occurring that causes this to happen with the API that could be slowed down or procedurally commanded to make this more smooth for the LimeSDR? Please advise - this is a critical issue for the success of SDRAngel to operate with the LimeSDR since there's nothing else out there that supports transmit as well as SDRAngel can do it.

f4exb, 2017-09-30:
If I get your scenario right I suppose that what you want to achieve is half duplex operation which should of course be possible with a full duplex device just by switching on and off the Rx and Tx sides in the correct sequence. 

It happens that I have a PlutoSDR connected and this works fine so I suppose it is a problem with LimeSDR. HackRF is not concerned as it is half duplex by construction. I haven't tried BladeRF yet in this half duplex scenario.

f4exb, 2017-09-30:
Well. I can mess up very easily by trying to start/stop on the Tx side whether the Rx is on or off...

It turns out that the code on the Rx side and on Tx side are different and the code on Rx side seems to have the start/stop functionality working correctly. On Tx it enables and disables the stream when doing start/stop where on Rx it is only at open/close.

f4exb, 2017-09-30:
I also encountered the same problem while stopping/starting the Tx while the Rx is running. The start/stop on Tx side is different because it needs to destroy and re-create the stream to actually stop transmission. 

It seems that suspending the other threads while doing the stream delete/create solves the problem. Will try this fix in the next release.

martywittrock, 2017-10-01:
Edouard,

Excellent, so happy that you observed the issue. This will make SDRAngel so much better in half duplex..! Please keep me advised when the release is posted so I can try it.

73 de Marty, KN0CK

f4exb, 2017-10-04:
Should be fixed in v3.7.3

closed 2017-10-04 by f4exb