f4exb/sdrangel #68: NCO can't be set to zero on LimeSDR

mntnmn, 2017-09-28:
I just rebuilt and now get the message "could not disable and set NCO to 0 Hz". It gets stray numbers stuck in there. I have upgraded LimeSDR firmware and LimeSuite also.

mntnmn, 2017-09-28:
I rebuilt LimeSuite, SoapySDR ans SDRAngel again with a sudo ldconfig at the end and I still get the same messages about the NCO but it seems to be working now.

f4exb, 2017-09-29:
SoapySDR does not come into play SDRangel uses LimeSuite directly. The latest version of LimeSuite (17.09.1) seems to fix a few bugs. Normally this sort of message is issued because of a bad return code from a LimeSuite function.

mntnmn, 2017-09-29:
I grepped LimeSuite and SDRAngel code for the messages and found one at line 63 in devicelimesdr.cpp.
The "could not disable and set NCO to 0 Hz" seems to be from limesdrinput but I haven't found it. I would guess that the device one triggered the input one.
It may be that LimeSuite has changed something again.

f4exb, 2017-09-30:
Actually a beter way to enable/disable the NCO would be using the CMIX_BYP_RXTSP flag like: 

    LMS_WriteParam(device,LMS7param(CMIX_BYP_RXTSP),1);

mntnmn, 2017-09-30:
This minor bug has gotten me to use the NCO all of the time. I just set the LO for something nearby and set the NCO to the interesting frequency. That way the DC is shifted out of view when I zoom in on the interesting frequency, but all of the offsets in the plugins stay zero.

f4exb, 2017-09-30:
It looks like using LMS_WriteParam would fix the problem.

f4exb, 2017-10-04:
Fixed in v3.7.3

closed 2017-10-04 by f4exb