f4exb/sdrangel #69: bladerf: swap I/Q?

wildy, 2017-10-01:
Hi.

I just started using bladeRF with SDRangel and noticed that when I'm changing the center frequency, signals actually go in the opposite direction, which is, AFAIK, the sign of that you'd need to swap I/Q.
Gqrx has this option and when I checked it I could see signals the right way, but in SDRangel I can't see such an option.
I would very much like to see this option in SDRangel if it's not implemented already. Thanks.

f4exb, 2017-10-02:
I use BladeRF regularly and I know that the frequencies go the right way. What you see are probably ghost images. There is no need for I/Q swap.

wildy, 2017-10-04:
Why then do I have a signal at, say, 433.125 but when I set the device frequency to 433.000 I see it at 432.875, while in Gqrx I can see it at its proper frequency?

f4exb, 2017-10-07:
Hello,

what are these settings in the BladeRF plugin GUI in this case:

  - sample rate (SR)
  - decimation (Dec)
  - center frequency position (Fc)
  - RF bandwidth (BW)

Also since the frequency is 433.125 I suppose you don't use the XB200 optional board. You must make sure it is not engaged in this case (xb200 combo).

In any case this is not an I/Q swap problem.

Best regards.

f4exb, 2017-10-21:
Closed as no reply from logger and non issue actually

closed 2017-10-21 by f4exb