f4exb/sdrangel #70: The way to limit selectable center frequency range

Satopp, 2017-10-03:
Hello,

 Capability of transmitting is excellent, I really love it, and enables to open more generic uses for ordinary people, not just experiments for skilled engineers. For such uses, it's better to have protection to avoid unintended off-band transmission by simple miss operation.

 Isn't there any simple way to set selectable frequency range? Will it be a enhancement request?

 Instruction to modify source code to build hard-coded specific executable would also be appreciated. 

Thanks in advance.

f4exb, 2017-10-03:
I am sorry but you have to know what you are doing when you transmit. This is what licences are for.

closed 2017-10-03 by f4exb

Satopp, 2017-10-04:
Let me have one more try.

I want to believe I know what I'm doing but I also make many mistakes as most people do. License doesn't assure to eliminate such mistakes. Simple protection is not perfect but certainly help to reduce such risks and worthwhile to have.

f4exb, 2017-10-04:
To answer your initial questions then:
 
  - "Isn't there any simple way to set selectable frequency range?": No
  - "Will it be a enhancement request?": I don't see the point to pollute the interface with this requirement. If you sit and think of it you will realize the risk of doing something really wrong like transmitting with significant power on the Air band disturbing vital communications is really really low
  - "Instruction to modify source code to build hard-coded specific executable would also be appreciated.": No way I am going to accept any pull request "hard" coding something. There's enough of it already that I am trying to move away.

The nearest thing I can think of that would be acceptable is the ability to load a specific user profile with particular frequency restrictions a bit like the regulatory file used by WiFi drivers. At this point there is no user management. It may become hotter when the web interface ramps up interfacing distant transponders or repeaters which is not going to be soon.

Satopp, 2017-10-04:
Thank you for sharing your view and thoughts on this. Understood well and it's now clear I need to give up.

f4exb, 2017-10-04:
I still retained the authentication idea via the web interface and possible frequencies restrictions attached to it and updated comments on #53 in that sense.

Satopp, 2017-10-05:
Thank you for your warm consideration.