f4exb/sdrangel #71: GUI closes without AM demod

mntnmn, 2017-10-07:
OS- Linux/Kali
LimeSDR, latest LimeSuite.
Latest SDRAngel
If I try to close an AM demod plugin the GUI closes with "QThread: Destroyed while thread is still running
Aborted". No other message.This happens even if I open two AM demods and close only one. It does not happen for any other RX plugin, only AM demod. If I try to open a preset that does not use AM it closes the GUI.

f4exb, 2017-10-08:
Correct: the proper delete sequence of m_channelizer and m_threadedChannelizer is missing in AMDemod::~AMDemod(). This used to be done in the GUI and was moved out in 3.7.3 to prepare GUI extraction.

mntnmn, 2017-10-09:
OK. It almost looked intentional. I was trying to find a change that would have caused it but was looking in the wrong place.

mntnmn, 2017-10-12:
I pulled and built the dev and this is fixed.

closed 2017-10-12 by mntnmn