f4exb/sdrangel #73: Delay is still present with LimeSDR on SDRAngel...

martywittrock, 2017-10-13:
I've been seeing an issue that when I first start a sink (transmit) session in SDRAngel the 7 - 10 second delay is still apparent, but as soon as I change the sample rate from something like 7.000 000 and then to 6.000 000 and then back to 7.000 000 the delay goes away and normal operation starts again for transmit voice.

Just adding this in since I've been doing A LOT more testing with transmit these days and it's great to have it...Just seems like there's one bug left on that.

73 de Marty, KN0CK

f4exb, 2017-10-14:
I suppose this happens when you start right after loading a preset. I see a bug that must be fairly old where the full reconfiguration is not always done after deserialization.

f4exb, 2017-10-15:
In addition the FIFO length is initialized in the thread so each time you start transmission it is reset at a fixed (usually too long) value.

f4exb, 2017-10-15:
Fix implemented in v3.7.4

martywittrock, 2017-10-16:
I have checked out 3.7.4.1 and it works spectacularly..!! No more delays on initial transmit - it appears that it's fixed..! SO HAPPY with this new Win64 version for the LattePanda - working GREAT!! 

73 de Marty, KN0CK

f4exb, 2017-10-17:
Tnx for reporting. Hence I can close the issue.

closed 2017-10-17 by f4exb