f4exb/sdrangel #74: 10 MHz Lock

n1jez73, 2017-10-15:
Is it possible to add the ability to use external 10 MHz to lock the LimeSDR  in SDRangel? We're looking at using the Lime on 3.4 GHz EME. JT65 transmit/receive is working, but accurate frequency stability would be a big plus!

I have used an external 10 MHz reference but it must be set up in LimeSuiteGUI and it's not permanent. It resets to internal reference with a power cycle.

Mike

f4exb, 2017-10-15:
I suppose this can be switched with the LimeSuite API in which case it should be possible. There's a small catch however because this is a global setting and should be reflected in all dependent plugins. This is also feasible.

f4exb, 2017-10-21:
http://docs.myriadrf.org/LMS_API/group___f_n___l_o_w___l_v_l.html#gaf654ad677321956d6d782a9003d939c4 it says “setting LMS_CLOCK_EXTREF changes clock source to external, use LMS_VCTCXOWrite() to change back to VCTCXO”

f4exb, 2017-10-29:
I posted a question in the LimeSDR forum: https://discourse.myriadrf.org/t/10-mhz-ref-input/813/8 Unless otherwise specified I will assume that the external frequency is selectable between 10 and 300 MHz continuously.

f4exb, 2017-11-05:
Implemented in v3.8.1

closed 2017-11-05 by f4exb