f4exb/sdrangel #76: CW keying envelope is "clicky"

kb1vc, 2017-10-18:
In looking at the keying envelope on a separate receiver, I see that there is no shaping on the leading or trailing edges.  This is generally not a good idea, as it creates additional sidebands and an annoying click in the ears of an operator.  Analog solutions usually put a crude RC filter in the keying line.  SDR code can do better.  SoDaRadio uses a cosine edge shape that looks like this: 

  float ang = 0.0;
  float ang_incr = M_PI / ((float) edge_sample_count); 
  for(unsigned int i = 0; i < edge_sample_count; i++) {
    rising_edge[i] = 0.5 * (1.0 - cos(ang));
    falling_edge[i] = 0.5 * (1.0 + cos(ang));
    ang += ang_incr; 
  }

Here is the keying waveform that I recorded from SoDaRadio/EttusUSRP listening to SDRangel on a LimeSDR.

[image]

I hope this helps. 

'73 
matt

f4exb, 2017-10-18:
Can you please detail how you got this result? There is indeed an envelope shaping function on the CW keying that zeroes both first and second derivatives that should perform better than the one you suggest. Here it is called "smootherstep". See: https://github.com/f4exb/sdrangel/blob/master/sdrbase/util/stepfunctions.h

kb1vc, 2017-10-20:
Eduoard, 

Thank you for responding.  SDRangel is a nice piece of work. 

How I observed the key clicks....

I put SDRangel in SSB mode, set the TX for 50.2 MHz and the tone offset to 1kHz.  Then I sent dahs at 13wpm with the TX gain set to 52dB.

The first hint was that it sounded "clicky" -- I generally let it go at a low element rate and listen on another receiver.  It sounds clicky on my FT817 and on my USRP N200/UBX. 

Then I looked at the spectrum in the waterfall.  There are two interesting aspects.  The spikes on either side of the elements indicate a sharp edge.  (Also note the carrier at 50.2 MHz.  The TX gain was set to 52dB.  This is not overload in the USRP, as I test it with input signals at this level all the time -- the line at 50.2 is the un suppressed carrier that is 30dB down from the intended signal.  Even at lower gains, the suppression is never much better than 30dB.)  

[image]

If I reduce the TX gain to 44dB the suppression still amounts to about 30dB.  The clicks are also still in the spectrum.  The signal into the USRP is well below anything close to full-scale, even when SDRangel TX gain is set to 50dB. 

I also see much larger "splats" at random intervals.  These may be caused by ALSA induced audio drop-outs, as such impulses often signal a sudden phase change in the signal.  

Finally, I used Audacity to look at the envelope.  Yes, it looks like it's got a 4ms attack and decay.  I think maybe 4mS might be a bit short.  I know that this puts stress on element shape at 26wpm, but at that rate, the information is in the clicks, and not the envelope.  It might be worth it to try a 15 or 20ms attack at low code rates.  (I notice that the dits at 13wpm are abou 90ms long.  I suppose that's right, but that would make a dot at 26wpm 45 ms long.  In which case I think a 10 ms attack would be just fine at the higher speed. 

The 4ms attack, not surprisingly, seems to be generating a click that is about 500 Hz wide -- I suppose if I spent a little time with Messr Fourier, a convincing argument could be made that 500 Hz is the right answer.  I would have expected a width of about 250 Hz.  It may be that the wider marks are from some bucket spreading, though I chose a Blackman Harris window in order to minimize the side lobes. In any case, the occupied bandwidth is going to be proportional to the edge rate on the envelope.  Going from 4ms to 10ms will cut the bandspread in half at least... This will also make the signal more gentle on the ears. 

The carrier suppression problem (which I think is the result of the Lime's problem with IQ balance) is the reason that SoDaRadio doesn't yet support the LimeSDR.  I was encouraged that you'd gotten the transmit side to work, and I expect I'll learn a lot from your trailblazing.

f4exb, 2017-10-20:
Thanks for the details. The dot length in seconds is derived from the formula 0.24 * (5 / WPM) which can be reduced to 1.2 / WPM and what is found in this document for standard timing: http://www.arrl.org/files/file/Technology/x9004008.pdf

At 26 WPM which is very fast and could be probably hardly used anyway we have a 46 ms dot so rise and fall times account for less than 20% so it could be made a little larger indeed. Another improvement would be to enlarge the ramp time as the WPM decreases.

4 ms is 250 Hz in frequency terms and may indeed have something to do with the 500 Hz splatter you observe. Twice the 250 Hz as one for each side or stg like that. If things are linear there could be some gain with the ramp time increase.

kb1vc, 2017-10-21:
Thanks Edouard.  Merci.

The reason I wasn't quite ready to say 4ms rise -> 250Hz or 500Hz is that there are a few rules that are floating around.  One says that BW = 0.35 / tr.  That's not a very good rule for predicting spectral content, it says something about the bandwidth required to carry the information in the rising edge.  Another rule is BW = 5/tr.  (See Howard Johnson and others.)  This is the bandwidth over which the transmission medium must be flat and well behaved in order to keep the eye of a bit stream reasonably open, assuming the bit time is about 5 * tr.  (It is a long and involved argument that has as much to do with economics and human behavior as it does with the Fourier transform.)  

In terms of measurable energy in the spectrum, there is likely real energy between 1/tr and 3/tr.  (Note that these are not harmonics, as the event is not periodic. There's something else at work here... The laplace transform of a raised cosine event smells like it has some promise of elucidating this.  Sure enough, when you get done with all the u(t) s and the cos(at + b) s and such, you end up with a sum that's got a s/(s^2 + (2pi/tr)^2) term in it. That's got a pole at 2pi/tr, and has some energy out to infinity, but the aperiodicity (the fact that it is only one edge over infinite time) contributes a delayed component that tends to reduce the higher spectral content a bit.   This is all from doing something that mathematicians get all woozy over (s = j \omega), but the real world allows us to do things that the blackboard just won't tolerate. ;)

Thanks again for listening.  This SDR stuff is fascinating.

kb1vc, 2017-10-21:
sorry, didn't mean to close it -- that's not my prerogative. ...

f4exb, 2017-10-21:
Sure, I will close it when v3.7.5 is out. As a preview this is what I get and will be soon in dev branch. At 12 WPM this is the dot linear amplitude shape. In fact the ramp eats up only 20% of the pulse since the falling edge ramp is after the dot period. The spectrum appears to be better behaved although there seem to be some spikes for an undetermined reason:

[image]

HackRF is used in transmission with the SSB modulator and receiver is a nearby RTL-SDR.

f4exb, 2017-10-22:
Implemented in v3.7.5

closed 2017-10-22 by f4exb