f4exb/sdrangel #77: demod data offset

mntnmn, 2017-10-18:
I have recorded a signal from the UDP source and found that for a very low signal I can see a negative offset to the data. It seems to be about 0x180 down, looking at the numbers in the file. This offset is even noticeable in larger waveforms. The signal should ride the zero line evenly plus and minus. I think it may be causing problems with some signal processing downstream as for a pure frequency the zero crossings aren't even.
Do you know where this offset comes from? For audio it has no effect.

f4exb, 2017-10-18:
> I have recorded a signal from the UDP source
Can you please detail? Do you record using file recorder and look at the values in a hex editor afterwards? UDP source has many options can you specify which settings you are using?

mntnmn, 2017-10-18:
It is NFM. I was thinking that the DC bias was from that. Aside from trial and error picking frequency settings, can you think of a way to eliminate this? For analog one just puts a capacitor in series.

f4exb, 2017-10-18:
If you want to eliminate DC with NFM there is another option and that is to use the NFM demodulator and its copy to UDP function (the "U" button). Address and port is set via the menu that opens by clicking on the small square on the top left corner (this is explained in the readme of the plugin).  NFM demodulator has a 300 Hz highpass filter to filter out the possible audio subtone hence the DC as well.

With the UDP source keeping the DC is intentional for digital modes over FM that need the DC component. 

If the NFM demod filter does not suit your needs I can make an update on UDP source NFM mode to add an optional DC filtering with a lower corner frequency.

mntnmn, 2017-10-18:
I'll give you're suggestion a try...

f4exb, 2017-10-18:
No... gain is fixed and sample rate also at 48 kHz.

mntnmn, 2017-10-23:
This works for me.

closed 2017-10-23 by mntnmn