f4exb/sdrangel #79: Can't enable Audio

FFY00, 2017-10-20:
I seem not to be able to enable the Audio in Preferences->Audio. I'm using sdrangel compiled from source and up-to-date with the latest master branch.

f4exb, 2017-10-21:
You should be able to see something like this:
[image]

Make sure ALSA is working properly and that you have pulseaudio set correctly. See: https://github.com/f4exb/sdrangel#notes-on-pulseaudio-setup

KC7NOA, 2017-10-29:
i have the same issue with Armbian ....

fresh install , recent kernel ... sound on web pages work ...
pulseaudio 8.0
i just get the default listing and no other devices

KC7NOA, 2017-10-30:
Gateware version mismatch!
  Expected gateware version 2, revision 11
  But found version 2, revision 8
  Follow the FW and FPGA upgrade instructions:
  http://wiki.myriadrf.org/Lime_Suite#Flashing_images
  Or run update on the command line: LimeUtil --update

Reference clock 30.720 MHz
RX LPF configured
AudioOutput::start: using default device 
AudioOutput::start: 48000 Hz S16_LE audio format not supported. New rate: -1
AudioOutput::start: Audio device (  ) failed

f4exb, 2017-12-06:
Non issue

closed 2017-12-06 by f4exb

FFY00, 2017-12-08:
This is a issue. I have everything set up as it is supposed to and I don't have sound.

jctual, 2018-01-21:
same problem for me.
No sound on Ubuntu 17/10
Works perfectly with other applications but not with SDRangel.
I tried all inputs proposed in Preferences/audio menu.
Yes this is an issue !

voxmeus, 2018-04-27:
Install pavucontrol. It is included in most distributions for example:
Ubuntu/Debian: sudo apt-get install pavucontrol
openSUSE: sudo zypper in pavucontrol

Open pavucontrol and open the last tab (rightmost) called 'Configuration'
get sound set up here, select your signal source, select the type of demod, then press the '+' button to add the channel, click the Go button and you should hear sound when you have a signal.