f4exb/sdrangel #80: Question, but possible feature request

th0ma5w, 2017-10-20:
Hello! Fellow, but very inexperienced, SDR hobbyist here... I was wondering if you had any ideas if SDR Angel could support integration with software like FLldigi https://en.wikipedia.org/wiki/Fldigi FreeDV https://freedv.org/ or WSJT-x https://physics.princeton.edu/pulsar/k1jt/wsjtx.html ... The commonalities seem to assume a physical radio, various interconnect options with regards to PTT or VOX, and soundcards for input and output.

Sound routing could perhaps be done with things like Jack, Virtual Audio Cable, or something, but PTT or VOX would need either some kind of serial port input that accepts either basic RTS monitoring, or even more complex HamLib style commands to support tuning... and VOX would just need some kind of sound card input level threshold monitoring in order to trigger TX.

I have a HackRF so it would be interesting to get running on QRP digital modes with it... Any other thoughts? I know a little C, but may be able to munge through some C++ if I tried... your code base looks very very clean and approachable.

f4exb, 2017-10-21:
Hello,

> Sound routing could perhaps be done with things like Jack ...

I am not going to support this in any way. The preferred way is to use UDP transfer instead. If 3rd party software cannot get input from UDP well... that's too bad.

FreeDV should get a plugin in its own right or integrated to DSD demod for FM mode. Ideally this would be the same for any other mode or group of modes: a plugin in its own right (or a pair for Rx/Tx). Of course using another piece of software would be convenient to functionally replace a missing piece but I am not going to actively support any other transfer than UDP.

For PTT in slave mode that is activated from another software I'd rather have it as a control via a web API. I opened #53 some time ago. It needs a large code refactoring which is in progress. This is also a pre-requiste for a headless variant.

> but PTT or VOX would need either some kind of serial port input ...

That's very complex, convoluted and just messy to integrate within SDRangel this is why I'd rather leave it to the users to build their own scripts or piece of software and eventually activate PTT (that is Rx off / Tx on and the opposite on the appropriate source and sink plugins) via a web API.

mntnmn, 2017-11-13:
I have a script to start wsjtx and get input ant output to SDRangel with pulseaudio.

#!/bin/sh
pactl module-load module-null-sink
pactl module-load module-loopback
wsjtx
pactl module-unload module-null-sink
pactl module-unload module-loopback

Start this script then SDRangel. Select null and null.monitor from the audio menu in sdrangel and wsjtx and then start streaming. There is no rig control so you have to do that manually. Maybe soon we will have external control over frequency and tx/rx. Right now I can use mine full duplex with LimeSDR.

f4exb, 2018-03-24:
Implementation of issue #145 can probably give the flexibility on the audio output that was required in this ticket

f4exb, 2018-03-30:
implemented in v3.14.0

closed 2018-03-30 by f4exb