f4exb/sdrangel #82: Obtaining Different Sampling Rate in TX than RX...

martywittrock, 2017-10-23:
Edouard,

I am finding that 7.000 000 S/s is perfect for SDRAngel in receive (Source) but I am also finding that TX (Sink) the audio quality of the transmitted voice sounds MUCH better if I am running a sampling rate of 3.000 000 S/s. Since the Sampling rate is coupled together, when I shift to a lower Sampling rate is Sink (transmit) my receiver is then tuned off it's original frequency (say, 14.080 MHz) but the Sink performs admirably there.

Is there a way to have independent Sample rates for Source and Sink or am I using SDRAngel wrong for receive and transmit to obtain the best audio quality in transmit and still hold my receive frequency?

73 de Marty, KN0CK

f4exb, 2017-10-23:
Hello Marty,

looking at LMS200M datasheet here: http://www.limemicro.com/wp-content/uploads/2015/09/LMS7002M-Data-Sheet-v2.8.0.pdf it appears that there is a single "clock PLL" block that drives all ADCs and DACs. Now I may have possibly interpreted this wrongly in that host/device sample rates should be the same on Rx and Tx. Of course this would work if interpolation and decimation factors are the same but in fact they are independent. So if I choose different factors for Rx and Tx it is quite possible that things go very wrong here. I will have to check this.

So if indeed the ADC and DAC clocks are the same with possible different interpolation and decimation factors then the host/device sample rate could be different in Rx and Tx but still be tied by the ratio of their interpolation and decimation factors. The ratio is not arbitrary so it will be a power of two in the 1 to 32 range(1, 2, 4, 8, 16, 32) so having 7 MS/s for Rx and 3 MS/s for Tx is not possible anyway but you could have 6 MS/s and 3 MS/s or 7 MS/s and 3.5 MS/s as the nearest options.

martywittrock, 2017-10-23:
Edouard,

That would be fine to have an equal ratio between receive and transmit - 7MS/s Source (receive) and 3.5MS/s Sink (transmit) would be good. I'm just finding that when I have the sampling rate (using 7MS/s) set higher in transmit the audio gets garbled or distorted, but when I run transmit at 3MS/s it sounds nearly perfect. My issue is that when I was trying to set up a digital mode (RTTY) over the weekend and had the proper sideband selected (and listening to my signal on my FT-757 Yaesu rig) the audio was quite distorted and then I tried just straight voice and it was also distorted. But as I adjusted the sampling rate down in transmit until I arrived at 3MS/s it began to sound more normal and less distorted.

Anything you can do with this would be greatly appreciated, Edouard. SDRAngel is getting A LOT of attention with every post I've been submitting to MyriadRF as well as Youtube as the application of choice for the LimeSDR for Amateur Radio. 

73 de Marty, KN0CK

f4exb, 2017-10-23:
I didn't even realize before but I suppose we are talking about LimeSDR here. I updated the title to reflect this.

martywittrock, 2017-10-23:
Edouard,

Yep - the finding is for the LimeSDR - thanks,

Marty

f4exb, 2017-10-23:
If I read the note here correctly http://docs.myriadrf.org/LMS_API/group___f_n___a_d_v_a_n_c_e_d.html#ga6e8212534d71682a77b6c0a2c8d201bb then indeed Rx and Tx rates can be different and related by a power of two. So I think the present implementation is not working generally only if decimation and interpolation factors are the same.

martywittrock, 2017-10-23:
Edouard,

So if I change the interpolation and/or decimation values for each (Sink/Source) then I will be able to have separate Sampling rates? Give me an example that will best fit a scenario where the Sink (transmit) Sample Rate is 3.5 MS/s and the Source (receive) Sample Rate is 7.0MS/s. Provide what the Interpolation or Decimation need to be set at to achieve this.

Looking forward to this information to understand better - 

73 de Marty, KN0CK

f4exb, 2017-10-23:
According to LMS documentation yes you should be able to have different host to device (Tx) and device to host (Rx) rates because the ADC and DAC are constrained to have the same sample rates but there are independent decimation and interpolation factors. But presently in SDRangel the host to device and device to host rates are constrained to be equal which is incorrect and will work as expected only if you have the same decimation and interpolation factors.

A working example would be the following: let's say you set the Rx device to host sample rate (the one the USB link sees) to 7 MS/s and you have a decimation factor of 8 then the ADC sample rate is 56 MS/s so on the Tx side the DAC sample rate is the same. If you set the interpolation factor to 16 on the Tx side then the host to device sample rate will be automatically set to 56/16 = 3.5 MS/s.

At the moment this is not working like this and if you set the Rx to 7 MS/s then the Tx part will think it is also running at 7 MS/s where in fact it is running at 3.5 MS/s so the result is unpredictable and most probably wrong therefore I think this ticket is a bug fix rather than an enhancement.

f4exb, 2017-10-23:
After some testing the Lime does not seem to be working with different interpolation and decimation factors so in fact I will have to tie them together also. So sorry Rx and Tx frequencies will always be the same.

f4exb, 2017-10-23:
It seems concurrent Rx and Tx does not work anyway

f4exb, 2017-10-24:
Oh-oh! applySettings in the start/stop sequence has nothing to do here maybe this is causing trouble.

f4exb, 2017-10-24:
Issue #83 opened to track Rx/Tx concurrent updates problem

martywittrock, 2017-10-24:
Edouard,

EXCELLENT..! Thanks for working that in..! SDRAngel just keeps getting better and better with every release...BEST app out there for receive and transmit for the LimeSDR..! Keep me advised on your progress there - 

73 de Marty, KN0CK

f4exb, 2017-10-25:
Things do not work exactly as expected I have to figure out the actual arithmetic.

f4exb, 2017-10-28:
Done in v3.7.6

closed 2017-10-28 by f4exb