f4exb/sdrangel #84: AGC for NFM / AM

texasyojimbo, 2017-10-29:
AGC for SSB demodulator was added as enhancement in release 3.5.3 (issue #40 ).

Would be nice to have this for NFM also, and possibly also for AM.

f4exb, 2017-10-29:
NFM does not need AGC since we're interested in the phase of the signal.
I have to check for AM. I though there was some kind of AGC already.

texasyojimbo, 2017-10-29:
OK, perhaps it's just a PEBKAC issue. It seems like when I am using NFM the audio is distorted and overdriven, even when I adjusted RF and audio BW and signal gain.

f4exb, 2017-10-29:
The RF bandwidth is tied to the expected FM deviation by a static correspondence. This is the parameter you should tinker with in FM. It is working OK for me on most signals and I use NFM quite a lot.

texasyojimbo, 2017-10-29:
Thanks. I have also identified part of the problem as the DC spike. I was listening to APRS traffic on 144.39 and when I filtered out the DC spike, it stopped sounding so strange.

f4exb, 2017-10-29:
There is an AGC with AM and the NFM issue was related to the DC spike so I think this issue can be closed

closed 2017-10-29 by f4exb