f4exb/sdrangel #86: SDRAngel V3.7.8 has issue with Sampling Rate...

martywittrock, 2017-10-31:
Edouard,

I tried the latest 3.7.8 SDRAngel and it seems to have lost the ability to have separate Source (receive) and Sink (transmit) sampling rate capability like was in 3.7.6-1. I'm having enormous problems setting up (for example) Source at 7.0MS/s and Sink at 3.5MS/s without the other being affected. I was adjusting the Source to 7.0MS/s and then went to Sink to adjust it to 3.5MS/s and came back to find the Source at 80MS/s..!! I adjusted the Source back to 7.0MS/s and then the Sink was at 0.875 MS/s. It seems they are coupled again - not independent like they were before, and I was depressed to see this. So I went back to 3.7.6-1 and everything was fine again.

Hope this helps...It was working SO WELL in 3.7.6-1, too...

73 de Marty, KN0CK

texasyojimbo, 2017-10-31:
Another thing I noticed tonight in 3.7.8 is that RX1 and RX2 seem to flip-flop. In prior releases, the first source always seemed to map to channel 1 on the LimeSDR (RX1) and the second to channel 2(RX2). 
It was my understanding was that this was by design. 

However, I was playing around with some directional couplers just now (so I am paying very close attention to which channel is which as it maps to which one is forward/reverse power!) and it seems that R0 was being mapped to RX2 and R2 to RX1.

It's possible this is just a PEBKAC (again) on my part, but it might also be related to KN0CK's point which is why I raised it.  It's possible that I started streaming R2 before R0, and maybe that's when the mapping occurs, rather than when the source/sink tab is created?

It seems to me that it would be more intuitive to the user to just present channel 1/2 to the user directly in the GUI somewhere possibly under antenna selection.

texasyojimbo, 2017-10-31:
I've confirmed with further experimentation that RX1 gets assigned to whichever source device tab has the LimeSDR device assigned (checkmark clicked) first under "Sampling devicss control." As such my previous comment is confirmed as PEBKAC/user misunderstanding. Disregard my comment as bug but perhaps note as a UX/UI enhancement request?

I can also confirm Marty's point regarding sampling rates. For example when I change samplle rate on T1 to 2 MHz, sample rate on R0 seems to change to 4MHz from 5MHz automatically. I also notice some weirdness relating to decimation settings...

f4exb, 2017-10-31:
What you have to realize is that Rx and Tx sample rates are not independent by hardware limitations. In the LMS7002M there is a single clock source for the DAC and the ADC. The hardware still allows for ADC and DAC sample rates to be related by a power of two that is DAC sample rate is twice the ADC or vice versa or they can be equal. The exact way this is done is not up to the client application this is all done in LimeSuite. 

My code has not changed regarding the sampling rate and hardware decimation and interpolation algorithms between 3.7.6 and 3.7.8 the only change that can be related is that 3.7.8 is compiled with LimeSuite 17.10.0 while 3.7.6 was with 17.09.1

I am not going to stay stuck with LimeSuite 17.09.1 so the only way forward is to ask the Lime guys to enhance their interface although I vaguely remember that the ADC to DAC ratio can be configured in LimeSutie GUI so there could be a chance that there is some API function to guide the ratio calculations.

As a possible workaround you can set the ratio between hardware decimation and interpolation factors to more than 2 (i.e. 4, 8, ... at least 4) then the resulting host sample rates will be related by at least a factor of two.

The other issue is not related and will be indeed followed on issue #52

texasyojimbo, 2017-10-31:
It looks like 17.10.0 includes/relies upon the 2.11 gateware which makes some kind of change regarding sample rate on different channels. However I don't understand the change. Marty -- maybe we should  ask MyriadRF?

f4exb, 2017-10-31:
I upgraded to 2.11 gateware and it seem it works as it used to be with 3.7.6.

f4exb, 2017-10-31:
I found the details on the ADC and DAC clock generation on page 17 here: http://www.limemicro.com/wp-content/uploads/2015/09/LMS7002M-Data-Sheet-v2.8.0.pdf
"There is a fixed divide by 4 within the ADC block hence clock division
on the DAC side to provide more flexibility. There is a MUX to connect
either Fpll, or Fpll/M to either ADC or DAC clocks. M is programmable
and can be set to M = 1, 2, 4 or 8. ". Therefore:

  - the DAC sample rate can be 4 times or twice the ADC sample rate
  - sample rates can be equal
  - the ADC sample rate can be twice the DAC sample rate

And this is supposed to be "programmable"

f4exb, 2017-11-01:
This is too complex to control the CLKMUX via the low level API so unless the Lime guys provide a high level API we will have to stay with the default behavior.

martywittrock, 2017-11-01:
Edouard,

I reprogrammed my LimeSDR last night to the latest gateware and observed on 3.7.8 that it seems to run fine at 7MS/s for the Source (receive) but I cannot adjust the Sink (transmit) to half the value (3.5MS/s) without crashing SDRAngel now. They have to be the same sample rate. I used to be able to tune them independently in 3.7.6-1 and it worked wonderfully, but with the new gateware if I even try to set the transmit lower than 4MS/s SDRAngel will just crash. Also, I noticed that when I went to adjust the span of the SSB modulator for Sink, as I put my mouse on the control and clicked to move it SDRAngel crashed, too. I went back to 3.7.6-1 and tried the same thing and it didn't crash - I was able to adjust the span normally and it worked. So there's something with that, too.

Let me ask this - For the LimeSDR for both Sink and Source, what are the settings you use for Sampling, H, and S? I have been trying to set my Sampling to 7MS/s and H=16, S=64 in both Sink and Source. But if these are not the right values to use for SSB voice (and eventually data) then I would like to know what settings you use to get clear undistorted audio. I have found that if I run a Sample Rate of 7MS/s in Sink that the voice is intelligible, but it's kind of gravely (has some distortion in it). But when I ran (in 3.7.6-1) 3.5MS/s the voice quality was excellent - and this was when I was running Source at 7MS/s which works wonderfully, too. Now with both at the same Sampling Rate (7MS/s) Source sounds and operates excellent and Sink works, but the modulation isn't as clear as half the rate (3.5MS/s). If there's another way to get more clear audio in transmit with a different setting of H and S then please let me know and I'll try that.

Just my observations with 3.7.8 as of last night - but I'm certain in time this will all get sorted out.

73 de Marty, KN0CK

f4exb, 2017-11-01:
Hello Marty,

it takes quite a bit of fiddling to get things the way you like since it is entirely up to LimeSuite to decide on the hardware factor and sample rate on one side when you change one of these on the other.

To get things reproducible you have first to start SDRangel select Lime as the Rx set the hardware decimation H: to 8 and exit. Next time when you open SDRangel it will load this default.

So on the next run:
  - leave the Rx at H: 8 and SR: 5 MS/s
  - add a Tx tab and select the Lime on Tx this will be H;2 SR: 5 MS/s but the Rx changes to H:2 SR: 10 MS/s
  - on the Tx tab chagne SR to 3.5 MS/s. Then the Rx is changed to H:2 and SR: 14 MS/s
  - change the Rx SR to 7 MS/s. The Tx stays at 3.5 MS/s
  - at this point the ADC rate is 14 MS/s which may be a bit low. You can increase H: to 4 or 8 on Rx side and this will not change the SR of 3.5 MS/s on Tx

Hope this helps,
Best regards, Edouard.

f4exb, 2017-11-02:
In fact I am afraid the 3.7.8 binary for Windows was not compiled with 17.10.0 but still with 17.09.1

martywittrock, 2017-11-04:
I am pleased to report that with my LimeSDR firmware updated to the most recent release, Sampling rates set at 5.0 MS/s Source and Sink and keeping the same settings for H and S (H=64, S=16) both transmit and receive I'm finding that the operation with 3.8.0.7 to be REAL good and without any abnormalities observed thus far. Transmit is clear and clean and receive is very responsive. I have more testing to perform, but I have to say for what I see thus far this looks like a GREAT release. I will post if I have other observations, but this is probably an even better release than 3.7.6-1.

f4exb, 2017-11-08:
Good news! So I think this can be closed now.

closed 2017-11-08 by f4exb