f4exb/sdrangel #88: RTL-SDR segmentation fault when adding WFM

cyrus-and, 2017-11-05:
### Reproduce:

1. select RTL-SDR device;
2. start acquisition;
3. add WFM demodulator.

Compiled from master on Debian Linux. I can add other details if needed.

### Stacktrace:

```
#0  0x00007ffff726d44a in fftfilt::create_filter (this=0x0, f1=-0.625, f2=0.625) at /home/cyrus/repos/sdrangel/sdrbase/dsp/fftfilt.cxx:106
#1  0x00007fffe17a4a8e in WFMDemod::applySettings (this=0x5555564fff90, settings=..., force=false) at /home/cyrus/repos/sdrangel/plugins/channelrx/demodwfm/wfmdemod.cpp:271
#2  0x00007fffe17a471b in WFMDemod::handleMessage (this=0x5555564fff90, cmd=...) at /home/cyrus/repos/sdrangel/plugins/channelrx/demodwfm/wfmdemod.cpp:205
#3  0x00007ffff72590f9 in DownChannelizer::applyConfiguration (this=0x5555564ff310) at /home/cyrus/repos/sdrangel/sdrbase/dsp/downchannelizer.cpp:189
#4  0x00007ffff7258ec3 in DownChannelizer::handleMessage (this=0x5555564ff310, cmd=...) at /home/cyrus/repos/sdrangel/sdrbase/dsp/downchannelizer.cpp:122
#5  0x00007ffff72802dc in ThreadedBasebandSampleSink::handleSinkMessage (this=0x5555564f6e20, cmd=...) at /home/cyrus/repos/sdrangel/sdrbase/dsp/threadedbasebandsamplesink.cpp:116
#6  0x00007ffff7266e35 in DSPDeviceSourceEngine::handleSynchronousMessages (this=0x55555588b1c0) at /home/cyrus/repos/sdrangel/sdrbase/dsp/dspdevicesourceengine.cpp:580
#7  0x00007ffff7296ebd in DSPDeviceSourceEngine::qt_static_metacall (_o=0x55555588b1c0, _c=QMetaObject::InvokeMetaMethod, _id=2, _a=0x555556500960) at /home/cyrus/repos/sdrangel/build/sdrbase/sdrbase_automoc.dir/moc_dspdevicesourc_LNHD2QFNSGLA64.cpp:78
#8  0x00007ffff5e9f499 in QObject::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff6755b8c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff675d341 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff5e729e0 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff5e7516d in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff5ec6c43 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff0eba7f7 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007ffff0ebaa60 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007ffff0ebab0c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007ffff5ec704f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007ffff5e709ca in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff5c9e0f3 in QThread::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff726524c in DSPDeviceSourceEngine::run (this=0x55555588b1c0) at /home/cyrus/repos/sdrangel/sdrbase/dsp/dspdevicesourceengine.cpp:64
#21 0x00007ffff5ca2da8 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x00007ffff2b7f494 in start_thread (arg=0x7fffcab37700) at pthread_create.c:333
#23 0x00007ffff50a0aff in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97
```

https://github.com/f4exb/sdrangel/blob/3ff2b68fa7cf2e89a198b0adf9871c87d0eda012/sdrbase/dsp/fftfilt.cxx#L106

```
(gdb) p filter
Cannot access memory at address 0x18
```

---

*(I think that [`RELEASEWITHDBGINFO`][1] actually does nothing different than default.)*

[1]: https://github.com/f4exb/sdrangel/blob/master/ReadmeDeveloper.md#build-options

f4exb, 2017-11-05:
Had a look at this. WFM Demod is the only plugin failing in this case. The reason is that the sequence of actions during construction is done so that applySettings method is called before m_rfFilter is built and applySettings needs to configure m_rfFilter and therefore it should be constructed. Moving m_rfFilter earlier or better moving much later the DSPDeviceSourceEngine::addThreadedSink call that triggers the message that eventually calls applySettings solves the problem

cyrus-and, 2017-11-05:
If it may help further another a segmentation fault occurred in similar circumstances when I tried to remove the plugin (by pressing the X), but I didn't bother to detail how to reproduce that since I though it was probably related to the original issue.

f4exb, 2017-11-09:
Fixed in v3.8.2

closed 2017-11-09 by f4exb