f4exb/sdrangel #89: GUI: doesn't remember windows position and size

Radiomix2000, 2017-11-05:
When you close and then reopen app, window size and position are reset.

f4exb, 2017-11-05:
Can you be more specific? I don't see any problem.

f4exb, 2017-11-06:
Maybe this could be implemeted for dependent widgets at least the source/sink? http://doc.qt.io/qt-4.8/qwidget.html#saveGeometry

Radiomix2000, 2017-11-11:
Strange. I have this problem on W7 x64.
When I start the application and just maximize the window and close application, then again start the app, and now the window is not maximized. I will try to capture the screen, if need for explaining.

gm8arv, 2017-12-21:
I see the same issue on Win-10/64.  I like to have the window slightly larger than the default, and on the right-hand of my two monitors.  Neither the position (monitor) or the size are remembered across invocations.

f4exb, 2018-10-21:
Tried the save geometry thing but this does not make any difference. Seems to be a Qt issue so it cannot be fixed.

PR #946 (2021-07-01): Save and restore main window position and size
This patch saves and restores the main window position and size between invocations of SDRangel. Fixes #89
commit d2814e2d (2021-07-01): Save and restore position and size of main window between invocations. Fixes #89
closed 2018-10-21 by f4exb