f4exb/sdrangel #90: RTL-SDR: cannot start sample source

f4exb, 2017-11-05:
In version 3.8.1: trying to start RTL-SDR pops up a "Could not start sample source" message and streaming does not start

f4exb, 2017-11-05:
It seems to be a problem with my local build however librtlsdr has evolved since last pulling (v 0.7.0) and bias tee control has been added.

f4exb, 2017-11-05:
See: https://github.com/f4exb/sdrangel/blob/master/librtlsdr/CMakeLists.txt
This option should be added in the cmake command when compiling librtlsdr:
`-DDETACH_KERNEL_DRIVER=ON`

The new functionalities of rtl-sdr will be implemented in a different issue

closed 2017-11-05 by f4exb

gaspar1987, 2019-10-05:
[image]
hello I have a problem with the rtl sdr could not start sample source.

gaspar1987, 2019-10-05:
to get the problem back?