f4exb/sdrangel #94: Issue observed with GUI/App lock-up or end with adjustments in Channel window...

martywittrock, 2017-11-10:
In version 3.8.1 of SDRAngel, I've noticed that if I am in the receive mode and attempt to adjust the BW control on the Channel Demodulator (currently I'm using SSB Demodulator) SDR Angel will lock up or completely end (stopping the application completely that I have to restart it to recover). I have no observed if this happens to the Sink (transmit) Channel but it seems that if I try to adjust from Upper Sideband mode to Lower Sideband mode using the BW slider control the application will just stop and I have to restart it completely to recover. I have not seen this in 3.7.6-1.

f4exb, 2017-11-11:
The infinite loop on SSB demod changes has been fixed in 3.8.1 actually. I haven't noticed this problem anymore at least with the Linux build.

f4exb, 2017-11-13:
From the positive feedback on v3.8.3  on MyriadRF forum I think this one can be closed.

closed 2017-11-13 by f4exb