f4exb/sdrangel #95: make impossible

fe8769a, 2017-11-12:
hello 
I leave my error in French 

I have this error when doing make command

 83%] Building CXX object plugins/samplesource/rtlsdr/CMakeFiles/inputrtlsdr.dir/inputrtlsdr_automoc.cpp.o
make[2]: ***  Aucune règle pour fabriquer la cible « /opt/install/librtlsdr/lib/librtlsdr.so », nécessaire pour « plugins/samplesource/rtlsdr/libinputrtlsdr.so ». Arrêt.
CMakeFiles/Makefile2:2626 : la recette pour la cible « plugins/samplesource/rtlsdr/CMakeFiles/inputrtlsdr.dir/all » a échouée
make[1]: *** [plugins/samplesource/rtlsdr/CMakeFiles/inputrtlsdr.dir/all] Erreur 2
Makefile:127 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2

previously I did a cmake with 

cmake ../ -DLIBAIRSPY_LIBRARIES=/opt/install/libairspy/lib/libairspy.so -DLIBAIRSPY_INCLUDE_DIR=/opt/install/libairspy/include -DLIBHACKRF_LIBRARIES=/opt/install/libhackrf/lib/libhackrf.so -DLIBHACKRF_INCLUDE_DIR=/opt/install/libhackrf/include -DLIBRTLSDR_LIBRARIES=/opt/install/librtlsdr/lib/librtlsdr.so -DLIBRTLSDR_INCLUDE_DIR=/opt/install/librtlsdr/include -DLIBMIRISDR_LIBRARIES=/opt/install/libmirisdr/lib/libmirisdr.so -DLIBMIRISDR_INCLUDE_DIR=/opt/install/libmirisdr/include 

I manually copy both files in /opt/install/mibrtlsdr/lib 

any help would be welcome

f4exb, 2017-11-13:
simple: /opt/install/librtlsdr/lib/librtlsdr.so n'existe pas

fe8769a, 2017-11-13:
Je me suis planté dans mon message précédent mais 
/opt/install/librtlsdr/lib/librtlsdr.so existe bien

f4exb, 2017-11-13:
My cmake command is:
```
cmake -Wno-dev -DDEBUG_OUTPUT=ON -DBUILD_TYPE=RELEASE -DLIBMIRISDR_INCLUDE_DIR=/opt/install/libmirisdr/include -DLIBMIRISDR_LIBRARIES=/opt/install/libmirisdr/lib/libmirisdr.so -DLIBAIRSPY_LIBRARIES=/opt/install/libairspy/lib/libairspy.so -DLIBAIRSPY_INCLUDE_
DIR=/opt/install/libairspy/include -DLIBBLADERF_LIBRARIES=/opt/install/libbladeRF/lib/libbladeRF.so -DLIBBLADERF_INCLUDE_DIR=/opt/install/libbladeRF/include -DLIBHACKRF_LIBRARIES=/opt/install/libhackrf/lib/libhackrf.so -DLIBHACKRF_INCLUDE_DIR=/opt/install/li
bhackrf/include -DLIBRTLSDR_LIBRARIES=/opt/install/librtlsdr/lib/librtlsdr.so -DLIBRTLSDR_INCLUDE_DIR=/opt/install/librtlsdr/include -DLIMESUITE_INCLUDE_DIR=/opt/install/LimeSuite/include -DLIMESUITE_LIBRARY=/opt/install/LimeSuite/lib/libLimeSuite.so -DLIBII
O_INCLUDE_DIR=/opt/install/libiio/include -DLIBIIO_LIBRARIES=/opt/install/libiio/lib/libiio.so -DCM256CC_INCLUDE_DIR=/opt/install/cm256cc/include/cm256cc -DCM256CC_LIBRARIES=/opt/install/cm256cc/lib/libcm256cc.so -DLIBDSDCC_INCLUDE_DIR=/opt/install/dsdcc/inc
lude/dsdcc -DLIBDSDCC_LIBRARIES=/opt/install/dsdcc/lib/libdsdcc.so -DLIBSERIALDV_INCLUDE_DIR=/opt/install/serialdv/include/serialdv -DLIBSERIALDV_LIBRARY=/opt/install/serialdv/lib/libserialdv.so -DLIBMBE_INCLUDE_DIR=/opt/install/mbelib/include -DLIBMBE_LIBRA
RY=/opt/install/mbelib/lib/libmbe.so -DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..
```
I am afraid I cannot help more here. It is working for me and there are no other problems reported.

fe8769a, 2017-11-13:
Need to copy libs from differents SDR from the specific GIT (rtl, airspy) 
Now I have errors in SDRPLAY

In file included from /home/XX/Documents/sdrangel/plugins/samplesource/sdrplay/sdrplaygui.h:27:0,
                 from /home/XX/Documents/sdrangel/plugins/samplesource/sdrplay/sdrplaygui.cpp:20:
/home/XX/Documents/sdrangel/plugins/samplesource/sdrplay/sdrplayinput.h:126:5: error: ‘mirisdr_dev_t’ does not name a type
     mirisdr_dev_t* m_dev;
     ^
plugins/samplesource/sdrplay/CMakeFiles/inputsdrplay.dir/build.make:66 : la recette pour la cible « plugins/samplesource/sdrplay/CMakeFiles/inputsdrplay.dir/sdrplaygui.cpp.o » a échouée
make[2]: *** [plugins/samplesource/sdrplay/CMakeFiles/inputsdrplay.dir/sdrplaygui.cpp.o] Erreur 1
CMakeFiles/Makefile2:2992 : la recette pour la cible « plugins/samplesource/sdrplay/CMakeFiles/inputsdrplay.dir/all » a échouée
make[1]: *** [plugins/samplesource/sdrplay/CMakeFiles/inputsdrplay.dir/all] Erreur 2
Makefile:127 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2

make was 83 % and now 90 %  , at least the last 10 %

closed 2017-12-16 by f4exb